kylecorry31/Trail-Sense #683: Sensor details / Cell signal: doesn't seem to work (permanently 'No signal')

jer194, 2021-05-16:
Hi again ...
'Cell signal' is a nice feature, but it doesn't seem to work on my device (willow, on Android 11, TS 2.2.3).
Am I missing a setting, or is it a general problem?
Thanks - Jakob.

kylecorry31, 2021-05-16:
Hi, on the Sensor Details screen (bottom of settings), does it show something other than no signal?

It works on my device, and according to the documentation I'm doing the right thing - it's up to the android manufacturer to populate the cell signal cache, so if it isn't working there isn't much I can do (populating the cache from Trail Sense was causing issues on some devices)

kylecorry31, 2021-05-16:
Oops, I see in your title it says sensor details as well

kylecorry31, 2021-05-16:
Maybe I can add a setting to add the force cache population back again

jer194, 2021-05-16:
Unfortunately I don't remember.
I only have a backup of 1.6.0, the feature hadn't been implemented back then I think.

kylecorry31, 2021-05-16:
Yeah, it wasn't implemented then - I'll add it as a setting so if people are having issues with it, they can disable it (I never actually got reports, but I believed an error I was seeing in the Google Play console was due to it - I could have been wrong though)

jer194, 2021-05-16:
Great!
I can check whether it is fixed, once the code is there.

kylecorry31, 2021-05-16:
Awesome, I'll post the APK anyway (I'll put it in the dropbox) - should have it ready in a few minutes.

jer194, 2021-05-16:
Awesome!

Btw, for 'Gravity': I think that value is locked by Android to the nominal value (9.8067...).
I have a 15 min montoring on it for many days, it never changes.
'Linear acceleration' (whether raw or not) might give more useful information.

kylecorry31, 2021-05-16:
Gravity is meant to be locked to that value (though when you move the phone it might change a little until you stop - depending on the accuracy of the built-in algorithm) - it is only being used to sense gravity, rather than any linear acceleration as it is used to determine phone orientation.

If I add the motion alarm (#679), then I'll need linear acceleration there

kylecorry31, 2021-05-16:
Here's the debug build - it is the "Trail Sense - Debug.apk"

https://www.dropbox.com/sh/gplqslml3iyq2ec/AACn0Arv3KRHAUF1wCusmoZ2a?dl=0

jer194, 2021-05-16:
Ok, checking ...

For the discussion about gravity:

[t.pdf](https://github.com/kylecorry31/Trail-Sense/files/6489108/t.pdf)

I think 'they' normalize the gravity components in a way that sqrt(x*x + y*y + z*z) always result in 9.806...
Otherwise the linear accelerations should be visible in the gravity graph.

kylecorry31, 2021-05-16:
Yeah, from my understanding of the Android Gravity sensor is that it uses sensor fusion with the gyroscope and magnetometer to determine how the phone's orientation is changes and then use a low pass filter on acceleration to retain only acceleration caused by gravity. The resulting value gets normalized to have a magnitude of 9.806, this is fine in Trail Sense, I normalize it to a unit vector in all the calculations I do, as I only care about the direction rather than the magnitude. The sensor details page shows the magnitude in case there is an issue with the Gravity sensor, which would then show up as not being 9.81

jer194, 2021-05-16:
All perfect now :-)

kylecorry31, 2021-05-16:
Thanks for checking! I'll publish the new version now.

closed 2021-05-18 by kylecorry31