kylecorry31/Trail-Sense #693: In the new version (2.2.3), the application does not stop completely if it is manually force closed and when not in use

Ananas-Kuchen, 2021-05-21:
After I updated to version 2.2.3 (f-droid). the app keeps running again after being forced manually many times. can't even be stopped by force

Is this normal or not? Previously I was still using version 2.1.3 and when I force close it manually, the application no longer runs by itself. 

This problem is bothers me. 

thank you

kylecorry31, 2021-05-21:
I believe android will start up the background services even if force stopped (they don't run all the time, just on a schedule) - I don't think this is something I can prevent from happening - and Google actually markets that as a feature. My recommendation is to disable the services you want off in TS (TS settings), or turn on the in app battery saver to stop them

I'm assuming this happened on earlier versions as well?

Ananas-Kuchen, 2021-05-21:
> I believe android will start up the background services even if force stopped (they don't run all the time, just on a schedule) - I don't think this is something I can prevent from happening - and Google actually markets that as a feature. My recommendation is to disable the services you want off in TS (TS settings), or turn on the in app battery saver to stop them
> 
> I'm assuming this happened on earlier versions as well, but I can check

I use a restrict background activity in the battery saver setting but it has no effect only in version 2.2.3

I have installed this trial sense application from the start of its release until version 2.1.3 and as long as I use this application it will definitely not be active anymore if I kill it in the app settings or because it is killed by Superfreezz. and in the settings I always turn off the application conditions for automatic startup. and use it if I need.

but this is the first time I know this application can stay active again and as if it can't be stopped in the 2.2.3 version update. I use the manual method by killing in the app settings, with superfreezz but the app quickly reactivates without pausing. this confuses me

Ananas-Kuchen, 2021-05-21:
I'm using Android 8.1

kylecorry31, 2021-05-22:
Hi, the background processes in Trail Sense are controlled by Google's WorkManager, this is designed to work when the app is force stopped. The intended behavior from my perspective is that TS will continue to run the background jobs configured by the user (weather monitor, backtrack, pedometer, and sunset alert) even after a force stop - if it was not doing this, then that is a bug (though Google's documentation says that this behavior may be inconsistent between manufacturers). To stop this behavior, you must turn off the background service you don't want to run from within Trail Sense. I didn't update the workmanager version with the last release, so I don't know why that would change.

Is there a particular service/notification that is starting up when you don't think it should?

Ananas-Kuchen, 2021-05-22:
> Is there a particular service/notification that is starting up when you don't think it should?

I guess no

> Hi, the background processes in Trail Sense are controlled by Google's WorkManager, this is designed to work when the app is force stopped. The intended behavior from my perspective is that TS will [...]

Well, If the application runs without being forced to stop because it is running its job I don't think there is a problem. 

Thanks for the explanation.

closed 2021-05-22 by kylecorry31

Ananas-Kuchen, 2021-05-26:
Can you send the test results when the application is killed whether it really stops working completely or returns to active on your phone between v2.1.3 with v2.2.3 only?

```
example test result:

1. v2.1.3 = after being killed completely stops running

2. v2.2.3 = after being killed will wake up again
```

For the last one, I want to know if what I experienced is the same on other phones or not

thank you

kylecorry31, 2021-05-26:
So on my device, it does not appear that the jobs restart after force stopping (I know that WorkManager says it is supposed to, but it also says that it is not guaranteed which is unfortunate). This is for both versions.

Are the jobs restarting for you (weather, backtrack, pedometer)? If they are, then that is the desired behavior as long as you had them turned on in the first place.

I know you said you are using Superfreezz (I'm not familiar), and I'm assuming that scans for all active processes. If you are using that to determine if TS is active, I know in 2.2.X I added Android Quick Tile support (the quick tiles in the notification dropdown). When I have the tiles in my quick actions (see below), Trail Sense does remain active after force stopping (Android keeps the tiles running - this is controlled by the OS). I found if I remove the tiles from the panel (second image), then Android does not restart the tile services (according to adb shell ps). 

As for what is active in Trail Sense with the tiles there: each tile has a single service that checks the state of backtrack/weather/pedometer (the setting only, not whether it is running) when the tile is visible (as in the panel is open and you can see the tile). In the background, these services are idle - waiting for Android to tell them that they are visible.

[image]

[image]

Top: Tiles are removed from the quick actions
Bottom: Tiles are in the quick actions
[image]

Ananas-Kuchen, 2021-05-27:
> So on my device, it does not appear that the jobs restart after force stopping (I know that WorkManager says it is supposed to, but it also says that it is not guaranteed which is unfortunate). This [...]

If I see the backtrack button and pedometer can be active but the weather monitor can't be active (if I add it in the main notification menu).

But if I repeatedly tap the toggle menu in the settings (where I can sort the order of the notification buttons to show in the main menu). weather, countdown and pedometer no active sign

Is that natural?

As far as I remember I never put a toggle (backtrack, pedometer and weather monitor) in the main notification menu and the button is always off + GPS is off + airplane mode + only wifi I use

If i install v2.1.3 (where there is no button in notification yet) the app can be stopped and doesn't run itself again after being killed.

There may be a bug with notification toggle button added in v2.2.3 ...... I don't know 🤔

...

...

...

...

...

I give up. So I decided to add a trail of sense v2.2.3 to the whitelist on superfreezz so that it wouldn't get killed. 😵

Problem solved. I won't complain anymore. thank you for replying to my question with so much kindness. 

Despite the problems I mentioned, I still like this app

Ananas-Kuchen, 2021-05-27:
because there is no app like this in the entire fdroid repo that i searched lol

kylecorry31, 2021-05-27:
I'm not sure I follow what you are saying about the rapid toggling of the buttons? On mine, if I toggle the buttons I see the weather/pedometer notification but nothing for backtrack (because I have notifications turned off for that - I would expect a very quick "updating" notification while they perform their job). 

I do have an issue that I'm working on for this release to hide the weather tile on phones without barometers #688 - not sure if this applies to your phone. 

I'm also working on revamping the settings in the next release - I can add a setting to completely disable the tiles (I believe this is possible, as I need to do it for #688), which should allow you to completely force stop the app.