kylecorry31/Trail-Sense #696: Add time convert tool

kylecorry31, 2021-05-22:
Seconds, Minutes, Hours, Days

kylecorry31, 2021-05-22:
This should also have a time between function: pick two times and get the time between them (days, hours, minutes, seconds and a version in the "to" units)

aw-bib, 2021-06-21:
Adding the display of [Julian Date](https://en.wikipedia.org/wiki/Julian_day) might fit in here nicely.

kylecorry31, 2021-06-21:
@aw-bib what would knowing the Julian Date provide (specifically for hiking or wilderness survival)? I use this value frequently in Trail Sense's astronomy calculations, but I'm not aware of a use of it other than for computation purposes

aw-bib, 2021-06-21:
> I use this value frequently in Trail Sense's astronomy calculations

Guessed that, that's why it might be one of the cheapest additions.

> I'm not aware of a use of it other than for computation purposes

Indeed, as soon as one needs time differences and stuff, JD would be the way to go. Some astronomical tables use it. Depends of course on your focus, how far it stretches into _astronomy_. Your ideas to add eclipses and stuff made me think "quite a bit" and then JD and sidereal time are spring to mind immediately.

kylecorry31, 2021-06-21:
Thanks! I think for now I'm going to leave that out in favor of having this tool calculate time differences, and the astronomy tab using this internally to perform astronomy calculations.

kylecorry31, 2021-07-04:
The time calculator features were moved into #765

commit 7b2b6f6f (2021-07-04): Add time converter

Closes #696
closed 2021-07-04 by kylecorry31