kylecorry31/Trail-Sense #719: Noon alert

kylecorry31, 2021-05-31:
An alert at exactly solar noon.

Not sure if this will be implemented, but it was a feature request that I will research the potential for.

VarunBatraIT, 2021-06-01:
I am advocating this issue as well as https://github.com/kylecorry31/Trail-Sense/issues/718 

At the moment I am using many apps when I go on trekking - Suntimes is one such appw. I want trekking to be fun since I had not-so-good experiences in the first few loner trekkings. I was in pretty bad shape so I decided to go not so far for other few consecutive trekkings. Anyways here is why noons are important any Noon alert isn't exactly Noon alert it is always X mins before Noon alerts or X mins before Sunrise alerts. 

I put a lot of alarms and not necessarily follow them all but this is always what I do before going out. Trekking must be fun and thus I wouldn't want to burn my water intake when it is worst. So Noon is the time when I take a full hour of rest and eat and drink. 

Early in the morning, I wake up depending on the tools and stuff I have - sometimes it is 10 mins before sunrise (actually Civil Twilight) sometimes it is 1 hour before sunrise (actually Civil Twilight). Make breakfast and lunch if not have ready to eat and then I just put on packing and other stuff. If I waste say 20 mins, it will make a huge difference since daylight means trekking and night means rest - as simple as that. 

App sunrise doesn't have X minutes before {Event} but sometimes I just remember - for instance if I know after sunset there will be 42 mins of daylight, I know what to do when sunset alarm blows.

So feature request is not exactly Noon alerts, Sunrise alerts (alarms) It is tweakable - X mins before or X minutes after. Sometimes I am on other side of the hill and I would use skymap to time my alarms with sun position using its Time Travel feature. I know I overthink that - I go alone so I think it is not harm to be overprepared:)

VarunBatraIT, 2021-06-01:
Update: Just found that they do have a feature called offset in a seperate app. (SunTimes Alarm)

kylecorry31, 2021-06-01:
Thank you for providing more details/use cases!