kylecorry31/Trail-Sense #743: TS Full Export

kylecorry31, 2021-06-17:
Allow exporting of the following:
- beacons
- notes
- inventory
- settings
- tides (optional since experimental)
- maps (optional since experimental)

Preferably into a single file, but multiple files are an okay fallback (maybe zipped)

kylecorry31, 2021-06-17:
There should be the ability to import this file via onboarding or within settings

s-kaczmarek, 2022-03-29:
Hi, seems like this feature was added to milestone 3.1.0 but to be honest, I can't find it in the options. How to export and import all the trail sense settings and data?

kylecorry31, 2022-03-29:
Hi, this feature was added then removed from milestone 3.1.0. It has not been implemented yet.

kylecorry31, 2023-08-29:
This feature is available in the nightly builds under Trail Sense settings

commit d55e801c (2023-08-28): Update backup/restore text

Closes #743
closed 2023-09-01 by kylecorry31