kylecorry31/Trail-Sense #745: Solar eclipses

kylecorry31, 2021-06-19:
- Searchable
- Optional as a list item
- Includes start, end, and peak
- Lists percent of sun eclipsed

kylecorry31, 2021-09-02:
Naive algorithm (slow) that might work:
- If moon is not new, no eclipse
- For every minute of the day determine the following
  - Sun and moon azimuth
  - Sun and moon altitude
  - Semidiameter of the moon (needs distance, and altitude)
  - Semidiameter of the sun (0.266563889)
  - Threshold = (sun + moon semidiameter)
  - If abs deltaAngle(azimuths) < threshold and abs deltaAngle(altitudes) < threshold -> Eclipse

Lee-Carre, 2021-12-27:
> algorithm

[Celestia](http://www.github.com/CelestiaProject/Celestia) has an eclipse calculation (both future & historical) tool (which actually computes on the host, rather than fetching pre-computed data from elsewhere), which might offer insight.

You may also be interested in John Walker's (of AutoDesk fame) [Quarter Million Year Canon of Solar System Transits](http://www.fourmilab.ch/documents/canon_transits/) project (which discusses the methodology of what took 2 months of compute-time) and some of the rest of the [Astronomy](http://www.fourmilab.ch/?topic=astrospace) section of his site.

closed 2023-03-17 by kylecorry31