kylecorry31/Trail-Sense #759: [Crash] crash when clicking on multiple items in lists

qwerty287, 2021-06-30:
TS crashes if you click on multiple items in any list (user guides, tools, beacons...)
Logs
```
Navigation action/destination com.kylecorry.trail_sense:id/action_action_experimental_tools_to_rulerFragment cannot be found from the current destination a(com.kylecorry.trail_sense:id/fragmentToolSpeedometer) label=FragmentToolSpeedometer class=com.kylecorry.trail_sense.tools.speedometer.ui.FragmentToolSpeedometer

java.lang.IllegalArgumentException: Navigation action/destination com.kylecorry.trail_sense:id/action_action_experimental_tools_to_rulerFragment cannot be found from the current destination a(com.kylecorry.trail_sense:id/fragmentToolSpeedometer) label=FragmentToolSpeedometer class=com.kylecorry.trail_sense.tools.speedometer.ui.FragmentToolSpeedometer
	at androidx.navigation.NavController.d(:11)
	at b.a.b.o.v.a.onClick(:5)
	at android.view.View.performClick(View.java:7273)
	at android.view.View.performClickInternal(View.java:7237)
	at android.view.View.access$3600(View.java:802)
	at android.view.View$PerformClick.run(View.java:27910)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:214)
	at android.app.ActivityThread.main(ActivityThread.java:7397)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:935)
```

Probably that's not an easy fix, but maybe with some try and catches? Or don't fix it because that's actually a very special crash.

kylecorry31, 2021-06-30:
Interesting, I never would have found that haha

I have gotten an error report on that, and the user didn't know how to reproduce it and it didn't happen again - thanks for finding this!

qwerty287, 2021-08-09:
@kylecorry31 it's not fixed in the settings

kylecorry31, 2021-08-09:
@qwerty287 thanks, I'll fix that in the next release - forgot that uses the nav controller as well.

commit e8bcfd10 (2021-07-08): Fix crash on multi click

Closes #759
commit 0903948f (2021-09-07): Fix crash on settings multi click

Closes #759
closed 2021-09-07 by kylecorry31