kylecorry31/Trail-Sense #767: [bug] magnetic metal finder has higher values than the max value

JensKorte, 2021-07-05:
The phone is a Galaxy S5 plus.

[image]

JensKorte, 2021-07-05:
Calibration has a max value of 125.

kylecorry31, 2021-07-05:
I originally intended calibration to be for only removing Earth's magnetic field as this is intended for use outdoors. I'm assuming you are using this indoors to get this result?

I'll consider allowing the calibration of arbitrary magnetic fields with higher strengths though

JensKorte, 2021-07-05:
> I'm assuming you are using this indoors to get this result?

Yes, you are right ;)

kylecorry31, 2024-01-17:
The metal detector tool's threshold was recently redesigned to be a relative threshold, so I think this is resolved now. If not, feel free to comment here and I'll open it back up.

closed 2024-01-17 by kylecorry31