kylecorry31/Trail-Sense #776: Precise time scale astronomy & weather zones barometer

clover1980, 2021-07-12:
I think it needed more precise to minutes time scale in Astronomy tab and weather zone labels in barometer to see exactly where pressure (for novices) like in circle IRL barometers from old times where they label pressure zones by type of weather for instant understanding (rainy, clear and etc).

If weather labels can be justified by author, the precise time scale in Astronomy are quite important because during several months observations i've noticed when sun and moon joined together i have the most active productive time periods, including noticed numerous sleep endings during that astronomy event. Right now it's impossible to see precise time on scale, i want to tune my sleep cycles for that event with alarm clock for better productivity (i don't require this function, precise time scale is enough for me).

kylecorry31, 2021-07-12:
Hi,

I will add a way to see the sun/moon positions at specific times in the day (touch/drag chart) - and am going to refocus this issue to just be about that. 

As for the pressure zones printed on many barometers, my research has shown that they are not to be used as an "at a glance" weather prediction, but rather to be used for change over time (i.e. dial moves toward rainy may indicate worse weather). Trail Sense uses this method of interpretation, replacing rainy with worsening (as the dial moving toward rainy doesn't always mean rain - it could mean cloudy, rain, snow, etc). As always though, if credible contradictory evidence is provided I can make adjustments to Trail Sense's algorithm.

kylecorry31, 2021-09-04:
This will have a slider that can be adjusted to change the time. It will default to the current time. 

Changing the slider will change the sun/moon azimuth and altitude to match the time chosen. 

Optional: Stacked horizontal bar charts indicating whether the sun is up/down and moon is up/down that show the current time across them (like the old astronomy page)

closed 2021-09-14 by kylecorry31