kylecorry31/Trail-Sense #785: Add more coordinate grid formats

kylecorry31, 2021-07-16:
Support coordinate formats used for maps across the globe.

aw-bib, 2021-07-27:
Maybe the pointers to the OS grid (as one of the gird formats) come in handy before they get lost (#788)

- https://www.movable-type.co.uk/scripts/latlong-os-gridref.html
- https://getoutside.ordnancesurvey.co.uk/guides/beginners-guide-to-grid-references/
- https://en.wikipedia.org/wiki/Ordnance_Survey_National_Grid

Would come in very handy in the UK.

Lee-Carre, 2021-12-27:
> coordinate formats used for maps

Not just maps, but also those which are clearer for human-to-human communication (such as over a low-fi voice channel, and/or have denser encoding to require fewer characters be read(-back) for the same precision of location compared to purely-numerical types (I forget the name of the very short-string one I'm thinking of)).

Plus, ones that should be promoted (because common!=quality).

For clarity, I am absolutely not meaning the proprietary horror that is ///what.three.words {shudders}; libre-only.

As for maps specifically; include clues as to which format a user might want.
Why? Well, I have a user-story for you.

Visited Australia (Victoria, near Melbourne, to see family) for a month back in 2010. Took a traditional GPS(-only) receiver (Garmin eTrex; the older grey-body ones, with the monochrome display) with me. Rather limited compared to what a modern nano-computer with various apps can do now, but it's what I had at the time. My thinking was that I could at least save the coördinates of the various places we visited, in order to easily find them on a map when back home. The logging & re-finding coördinates part was successful.

One day, we were doing some 4×4 adventuring, and become less-than-certain of where we were (to decide which route to take). So, instead of hopeful-gazing at the map like others, I fetched GPS receiver to get the coördinates.

The trouble was, the paper map wasn't really designed for using regular latitude & longitude. It expected you to use whatever grid-system is common in Australia. None of us knew (and the map itself was none too helpful) which grid system that was. GPS receiver had many options for output; I just didn't know which to choose (and didn't feel like brute-forcing it, at the time; non-tech folks wouldn't have been receptive to that tedium).

If I recall, I think I (much) later determined that it might have been a 4-letter one beginning with M (MGRS(?); I'd have to re-check).

At the time, had the UI given hints as to which were likely to be used in Aus, then there may have only been a few possibilities to try.

A usability suggestion given what modern software is capable of; by default, list only the notations which are applicable to where the user currently is (or where on the map is being displayed, or where the beacon is; you get the idea: make it context-sensitive), with the possibility to show-all anyway (to allow for weird edge-case usage).

kylecorry31, 2021-12-27:
Good idea, and thank you for the background! Here's a new issue to track that: https://github.com/kylecorry31/Trail-Sense/issues/1087

I've also edited my previous comment to remove the word maps as I didn't intend this issue to just map coordinate formats.

Lee-Carre, 2021-12-27:
> have denser encoding to require fewer characters be read(-back) for the same precision of location compared to purely-numerical types (I forget the name of the very short-string one I'm thinking of)

[Open Location Code](http://www.github.com/google/open-location-code).

Lee-Carre, 2022-01-01:
* [Marsden Square translator service](http://www.bodc.ac.uk/resources/products/web_services/msq2cov/)
* as described in [Jersey sea level and coastal conditions climate review](http://www.gov.je/sitecollectiondocuments/government%20and%20administration/r%20jersey%20sea%20level%20and%20coastal%20report%20final%20dm%2020170803.pdf) (page 71):
  * Jersey Transverse Mercator
  * local ordinance datum
    * this may relate to the Ordinance Survey grid reference system used in Jersey

kylecorry31, 2022-01-01:
@Lee-Carre I had originally excluded Open Location Codes because of their lack of use in hiking (https://github.com/kylecorry31/Trail-Sense/issues/59#issuecomment-871260227), but hadn't considered the ability to read back the code - I'll give it another review with that in mind

Lee-Carre, 2022-01-09:
Systems common (ubiquitous?) in China (because [government department](//www.wikipedia.org/wiki/State_Bureau_of_Surveying_and_Mapping) says so):
* [GCJ-02](//www.wikipedia.org/wiki/Restrictions_on_geographic_data_in_China#GCJ-02)
* [BD-9](//www.wikipedia.org/wiki/Baidu_Maps#Coordinate_system)

Lee-Carre, 2022-01-30:
[OSGB Grid Reference to OSM Map](//osm.mathmos.net/osgb2osm/)

Junkyard5031, 2023-09-10:
Open Location codes AKA Plus Codes support would be cool. It's a system created by google but it has an open license.

Here is one: `9C3XGV5P+V7`

They are short, simple to read thanks to the a limited selection of letters (only 20 alphanumeric characters), easy to write down or even say verbally to others.
They are also supported by most navigation smartphone apps like OsmAnd, Organic Maps and Google Maps.

Seems like the perfect coordinate system for offline use and for this app.

https://plus.codes/
https://github.com/google/open-location-code
https://maps.google.com/pluscodes/support/
https://en.wikipedia.org/wiki/Open_Location_Code

kylecorry31, 2023-09-10:
@Junkyard5031 thank you for the suggestion. I believe I also have some more info about that in one of the new feature idea issues, but it is nice to have that all consolidated here for easy reference, thank you!

My only concern about that is the code displayed in Google Maps will be incompatible because it uses the city name shortened format - but the code itself is still useful outside of that now that it has picked up a bit more popularity.

kylecorry31, 2023-11-22:
It was also recommended to look into sometime like What 3 Words, or some non-proprietary version of that.

aw-bib, 2023-11-22:
> My only concern about that is the code displayed in Google Maps will be incompatible because it uses the city name shortened format - but the code itself is still useful outside of that now that it has picked up a bit more popularity.

I may be wrong but at least on OSM And this Google-Stuff only works if I have an online connection. So I assume they need some of G web services to resolve their proprietary codes into real coordinates.

kylecorry31, 2023-11-22:
> I may be wrong but at least on OSM And this Google-Stuff only works if I have an online connection. So I assume they need some of G web services to resolve their proprietary codes into real [...]

If you use the shorted code (which contains the city name), that requires an internet connection. If you use the longer (less readable) code, it does not require an Internet connection. So the version that I would implement in Trail Sense is definitely less usable than the one displayed in Google Maps and the code from Google Maps would be incompatible with Trail Sense unfortunately (but it works the other way around - Google Maps could use the code from Trail Sense).

kylecorry31, 2024-12-27:
Suggested by a user: EGSA87 (more details needed)

ElonSatoshi, 2025-11-17:
Okay, after researching plus codes, I think I like them more than geohash

Plus code is basically like geohash, but:
- Uses less letters, omitting more letters that might look similar or look like numbers (s, which might look like a 5)
- Widely adopted thanks to Google
- Uses the plus sign as a delimiter, a sort of decimal point for anything more precise than 8 characters

kylecorry31, 2025-11-28:
> Well, if you don't hardcode plus codes, but make them a plugin, that's cool too! Maybe I can help make a geohash plugin

Thanks - once I have the plugin support added I'll post a link to the sample here.