kylecorry31/Trail-Sense #793: [Maps] SVG files as maps

qwerty287, 2021-07-17:
The map tool shows SVG files in the file selection, but if I try to import one, TS crashes. Solutions: hide them from the file selection (and don't show TS as a share option for them) or support importing SVGs.

kylecorry31, 2021-07-17:
I think it would be good to support SVGs, I'll research what's available in Android to turn them into Bitmaps - and if there isn't a good solution, I will restrict them for now.

qwerty287, 2022-01-07:
I just found https://bigbadaboom.github.io/androidsvg/, don't know if it can convert them to bitmaps, but it can render them into a special android view

qwerty287, 2022-01-07:
It can render to canvas: https://github.com/BigBadaboom/androidsvg/blob/1d43f779ff489a6d2f97d256befcaf87ca773523/androidsvg/src/test/java/com/caverock/androidsvg/RenderTest.java#L52

kylecorry31, 2022-01-07:
Thanks! I would have to see how it works with the Subsampling Scale ImageView library I'm using

kylecorry31, 2023-08-28:
This may also benefit from #1345

kylecorry31, 2023-09-04:
I'm going to close this issue as I don't see SVG being a valid map format - if anyone has any examples of it actually being used, I'll be happy to reopen this issue.

closed 2023-09-04 by kylecorry31