kylecorry31/Trail-Sense #804: Add the Morse app to the app list

cewbdex, 2021-07-20:
While the app is not completed, it could be useful https://f-droid.org/packages/rocks.poopjournal.morse

kylecorry31, 2021-07-20:
Thank you, I will review this app and consider adding it to the recommended apps list.

kylecorry31, 2021-07-21:
After reviewing, I will not be adding this to the recommended apps list (or other Morse code apps). This app does look good though, but it just doesn't satisfy the hiking or wilderness survival criteria to be featured in Trail Sense.

Here are my reasons:
- Trail Sense already supports SOS as a flashlight or whistle
- Other morse code words would not be useful in a survival situation (in my opinion)
- If there are other morse code words that would be useful in a survival situation and are well understood, I can add them to Trail Sense

closed 2021-07-21 by kylecorry31