kylecorry31/Trail-Sense #819: Parametric packs

kylecorry31, 2021-07-30:
Very early idea stage:

- you can have a pack template which has items that change amounts depending on days planned
- you can copy a pack with a number of days
- a formula will be applied to specified items based on the number of days

Ex. Item 1 might have a minimum of 3 and maximum of 10 and be calculated as 2n where n is the number of days.

No idea if I'm actually going to implement this as it seems complex.

aw-bib, 2021-08-19:
Maybe some sort of `metapackage` fits in here as well.

Say for a hiking trip I need a some base layer. For a sailing trip I need the same.  However, the foul weather gear will differ and I'd not pack the sea boots for hiking and probably not the walking sticks for sailing. So I could imagine a package `hiking` that consists of certain sub-packages that may be shared with other activities.

kylecorry31, 2024-12-27:
Not going to implement this.

Reasons:
- Frequency of use (ex. edge case) vs effort to implement

closed 2024-12-27 by kylecorry31