kylecorry31/Trail-Sense #827: Temperature estimation at elevation

FTno, 2021-08-01:
When this feature is started the Base elevation field is filled with "0.0". I would prefer to start with a blank field (or background text). When "0.0" is erased the light grey background text "Distance" appears. I think Altitude or Elevation would be more suitable.

The Base temperature field is also pre-filled (no background text here after erasing).

The Destination elevation field also has the light grey background text "Distance" (Altitude or Elevation more suitable - or blank).

The Base elevation unit is always "Meter" at startup (fine - because this is my default value from settings), but the Destination elevation unit is always Feet. So I have to change this to Meter every time.

kylecorry31, 2021-08-01:
Thanks for the feedback! I plan on having buttons to fill in the base elevation/temperature from the corresponding sensor rather than auto-filling on load before I make it non-experimental. 

I'll fix the destination elevation unit, I must have only defaulted the base elevation.

kylecorry31, 2021-08-01:
I fixed the hint text and units in my latest commit and will be closing this issue. The auto-fill feature will be tracked by #828

closed 2021-08-01 by kylecorry31