kylecorry31/Trail-Sense #858: Change suggested boil time to desinfect water

neutron-anywhere-exclusive, 2021-08-22:
Boil time for water to desinfect water can depend on a variety of factors such as type of pathogens and pressure.
Therfore it is recommended to boil water for 3 minutes to be safe.
I would suggeste to set default timer to 3 minutes or include an option in the settings to change the duration of the timer.

kylecorry31, 2021-08-22:
Hi, the recommended boil time below a certain altitude is 1 minute by the US CDC: https://www.cdc.gov/healthywater/emergency/making-water-safe.html

If you are above that altitude Trail Sense will suggest 3 minutes.

I can add a setting to always use 3 minutes, but if you can find counter evidence from a trusted source that suggests 3 minutes all the time, I can remove the altitude threshold

kylecorry31, 2021-08-22:
I added the ability to choose the time on the boil timer and am closing this issue.
[image]

It defaults to auto, which uses the altitude to determine time. It suggests 1 minute for altitudes under 1000 m (the guideline specifies 2000 m, but 1000 m is safer with pressure fluctuations) and 3 minutes for other altitudes (or if the altitude can't be determined)

closed 2021-08-22 by kylecorry31

neutron-anywhere-exclusive, 2021-08-22:
Thanks for addressing the issue.
You definitely gone a step beyond with the auto feature. Love it!
The 3 minute recommendation comes from people like David Canterbury, however researching this, I came across the WHO guidelines http://apps.who.int/iris/bitstream/10665/155821/1/WHO_FWC_WSH_15.02_eng.pdf 
No boiling time is specified.
So I have to agree that 1 minute boiling is already more then enough.