kylecorry31/Trail-Sense #928: Pedometer without persistent notification

kylecorry31, 2021-09-18:
Other fitness / pedometer apps do it, I want to research more how to let the pedometer run without a foreground service and then allow the user to choose to have the notification or not.

kylecorry31, 2022-01-24:
From my research, it appears that this needs to run as a foreground service (so needs to have a notification) - I will experiment with what happens if the user disables the notification from system settings (and recommend doing that if it works)

kylecorry31, 2022-01-28:
Disabling the notification from the system settings allows this to work and has been added to the FAQ

closed 2022-01-28 by kylecorry31

Lee-Carre, 2022-02-02:
> Disabling the notification from the system settings …

Doesn't that leave the foreground service at risk for being killed? To my understanding, the notification is what prevents this.

kylecorry31, 2022-02-02:
As far as I can tell it stays active (I let it run for 3 days, opening TS here and there though). I also have had the weather and backtrack channels disabled for close to a year, and didn't have issues - maybe Android just hides the notification from the user, but treats it as an active notification/foreground service?

I can do more research just to be sure though.