kylecorry31/Trail-Sense #934: [Research] Microphone anemometer

kylecorry31, 2021-09-25:
- [ ] Frequency filtering
- [ ] Wind speed calibration
- [ ] Wind speed estimation

kylecorry31, 2021-09-25:
Related to https://github.com/kylecorry31/Trail-Sense/issues/460

kylecorry31, 2022-01-16:
While this feature would be pretty cool - I don't foresee it having any practical use under TS' use cases. 

Using the microphone would give a rough estimate of the wind speed - this is something the user could figure out themselves by being outside where they would be taking the reading anyway (ex. slight breeze or strong gusts). 

I'll give this a bit more thought in case there are edge cases I didn't consider, but as of now I do not plan on implementing this feature in TS (I might do a separate app for it, because I think it's pretty cool).

Lee-Carre, 2022-01-26:
A different method, which might be better for direction, would be to measure offset from vertical suspension.

That is, from a known (calibrated) neutral position (which may not be square to the floor; some phone cases have a strap attached at a corner), the angle from vertical is proportional to wind speed. The direction (while compensating for orientation; the phone being spun around the axis of the cord from which it's being suspended) from vertical is the downwind direction.

The intensity would be averaged for you, by inertia (where as wind noise across a microphone will be highly variable).

kylecorry31, 2022-01-26:
@Lee-Carre very interesting - I will play around with that. Even if it doesn't end up in TS, it would be fun to try.

kylecorry31, 2024-06-09:
Reopening - I'm going to be adding the microphone permission to TS (for the comms plugin: #1540). I want to give this another shot.

closed 2024-12-27 by kylecorry31