kylecorry31/Trail-Sense #940: Material theme styling issues

kylecorry31, 2021-09-27:
- [ ] Some FABs have a yellow border
- [ ] Some quick actions have the wrong tint/background color
- [ ] Text on primary color buttons should be the same (probably white - I'll check the contract score first)

ebraminio, 2021-09-27:
>  Light theme causes system top bar to become blue

Its current default is,
```
        <item name="android:statusBarColor">?attr/colorPrimaryDark</item>
```

What would the intended behavior here? I thought somehow its default is good somehow.

ebraminio, 2021-09-27:
Feels like the previous behavior `<item name="android:statusBarColor">#40000000</item>`

ebraminio, 2021-09-27:
>  Some FABs have a yellow border

A screenshot for this one would be nice :)

kylecorry31, 2021-09-27:
[image]

Thanks! Here's a screenshot of the packing list (note the issue isn't on the navigation/beacon page) - I didn't have time to debug it at all this morning

ebraminio, 2021-09-28:
So this can be closed now I'd guess :)

kylecorry31, 2021-09-28:
@ebraminio awesome, thanks for the help!

closed 2021-09-28 by kylecorry31