kylecorry31/Trail-Sense #960: Signaling: Large text

kylecorry31, 2021-10-03:
Display large text on the screen (user enters what it says) so the phone can be used to signal at a distance. This issue is to research the usefulness of this at different distances / light levels.

kylecorry31, 2021-12-26:
This could also be used as a way to send text between devices:

Have a second tab for a QR code that can be scanned

Lee-Carre, 2021-12-27:
For those of us who like using (task-)specific tools, then please make it possible (viz via intents) for inter-process communication. Both for acquiring the results of a scan, and generating codes from (textual\|human-readable) input. Possibly also transmitting the result (though there's yet other apps for that specific (sub-)task).

I'm thinking specifically of [Binary Eye](http://www.github.com/markusfisch/BinaryEye), here.

kylecorry31, 2021-12-27:
@Lee-Carre good idea, here's the new issue to track that: https://github.com/kylecorry31/Trail-Sense/issues/1086

As for transmitting the result of the scan, I'm currently supporting location and web URL intents (with an intent chooser, so the user can choose the app to open them with).

closed 2021-12-27 by kylecorry31