kylecorry31/Trail-Sense #972: Add approximate location support

kylecorry31, 2021-10-10:
This might not be possible when using just the GPS - as far as I can tell the GPS is not available when approximate location is enabled. 

This issue is to research how the approximate location can be supported (and where it would make sense to support it - obviously not on the navigation tab). 

It might be that this issue would allow approximate location support: https://github.com/kylecorry31/Trail-Sense/issues/432

Lee-Carre, 2022-01-01:
How about simply querying the `passive` location provider? That way, if other apps (actively) queried for accurate location (e.g. a map navigation app) then this'll be available to TS.

kylecorry31, 2022-01-01:
Do you have documentation on that / can clarify what you mean? For reference, TS does not use Google Play Services, so uses the native Location class instead

Lee-Carre, 2022-01-01:
> Do you have documentation on that / can clarify what you mean?

* [Android Location Providers (GPS, Network, Passive)](http://www.developerlife.com/2010/10/20/gps/)
* [Android Developers § LocationManager](http://developer.android.com/reference/android/location/LocationManager)
* [martykan/forecastie#611 (comment)](http://github.com/martykan/forecastie/issues/611#issuecomment-1001077202) (& subsequent comments)

> TS does not use Google Play Services, so uses the native Location class instead

To my limited knowledge of the internals, I think this shouldn't be a problem. I think it's part of Android (the OS) rather than Google Play Services.

I use other apps which I'm pretty sure are free of Play Services, but which use the `passive` location provider.

kylecorry31, 2022-01-01:
Thanks - I was not aware of this / thought it was a Google Play Services feature, but from the docs it looks like it isn't. I'll consider this when addressing this issue.

Lee-Carre, 2022-01-03:
See also [android/location-samples](http://github.com/android/location-samples).

kylecorry31, 2024-12-27:
Not going to implement.

Reasons:
- Frequency of use (ex. edge case) vs effort to implement
- High maintenance burden / likely to cause confusion
- Not technically feasible / will be inaccurate

closed 2024-12-27 by kylecorry31