kylecorry31/Trail-Sense #999: Battery drain?

kylecorry31, 2021-11-07:
I've had a few reviews on Google Play that say that Trail Sense is draining battery at an alarming rate. Unfortunately, none of the users have responded to me, so I can't identify what is the cause. I can't replicate this issue - the closest I can get is setting backtrack to 1 minute intervals, but that doesn't drain battery that quickly (and when a user sets that small a frequency, it alerts them it can cause battery drain).

Some of my thoughts as to the cause are:

- Users set a small interval for backtrack
- Users enabled the pedometer and it is not optimized by the device manufacturer
- There is a bug (ex. GPS/sensor polling is not cancelled and it gets stuck running in the background)
- Users are talking about using Trail Sense in the foreground (screen on) - not sure if there is much to do there, other than reducing GPS frequency on navigation (the screen is the main energy draw)

Lee-Carre, 2021-12-07:
Perhaps they mean that Trail Sense does no power saving when the OS is set to battery-saver mode?

While I'm all for efficiency, beyond a point there's not much you can do (especially if none of those experiencing the problem respond to your queries). The problem may not even be with Trail Sense itself; it could be a bug in an old version of Android. Does Google offer any clues about the environment which the users which experience this problem are using?

I do map ([OSM](http://www.openstreetmap.org/)) surveying, sometimes for 12-hour sessions, and so I simply accepted that for apps to work well, have many running, and all for multiple hours, I definitely needed an external battery. Without one, my device's internal (~3Ah) battery depletes at about 1%(age point) per minute.

I think there may be some merit in your suspicion that much of the draw might be due to the display. I find that this makes a significant difference (on the order of several Ah) in energy-usage during a day.

Unfortunately I don't use Trail Sense enough (but, I like it very much) to have experienced any extreme battery depletion described by others. So, can't add more in terms of a resolution.

kylecorry31, 2021-12-07:
@Lee-Carre thanks for your response!

Given there are so few reviews on Google Play regarding battery drain, I can't really tell if there is a trend (at first glance there isn't - an LG and Samsung phone running Android 9 and 11 respectively). 

I do feel like it is the screen causing the drain - if you are navigating using the compass, you might leave the screen on since there is no background navigation (plus the extra battery usage from GPS and compass sensors). Until I hear from a user who is willing to provide the details I need to help debug it, I'm just going to continue to make small code optimizations (as I do every release), which will reduce any excess battery drain.

I could also explore more options around the power-saving mode. There is a setting to turn it on automatically when the system indicates there is low battery/apps should conserve energy - though it is off by default.

Lee-Carre, 2022-02-13:
> if you are navigating using the compass, you might leave the screen on

On the assumption that this is a factor; for AMOLED displays a darker theme will reduce energy usage. I notice that a (dark\|night) theme is already implemented, now (too many relevant tickets to enumerate them). So, perhaps a widget to temporarily enable a dark theme (or, more accurately / specifically / applicably, a high-contrast theme as per #293), for situations of wanting an always-on reference while minimising the Jules(?)

If possible, dim the display, too? Restore previous brightness upon input (tap) or a significant UI event.

Lee-Carre, 2022-02-13:
> there is a toggle to enable power saver mode (quick action, settings, and battery tool) which turns on the "Black" theme.

> Is that along the lines of what you were suggesting as a widget?

Essentially, yes. That covers functionality. However, it may not be apparent to users that this possibility exists.

I still think that only invoking power-saver mode, when the battery is already mostly depleted, is suboptimal. If it's enabled preemptively, then the run-time is significantly increased.

> I'm not sure I would want to [dim the display] — since that relates to the system rather than Trail Sense.

Indeed, I had reservations about trying to change system settings. Though, it could be done on a theme level within TS; darker colours.

kylecorry31, 2023-08-28:
I have not received any more reviews / questions about battery usage - nor have I noticed an issue myself (I run it 24x7). I'm going to close this, and if a user is able to provide a specific cases where it drains excessive battery, I will create another issue. 

There are other issues related to stopping all services and recording wakeup times which should help (in both power consumption and diagnosis)

closed 2023-08-28 by kylecorry31