kylecorry31/Trail-Sense #1009: Upgrade to Material3

kylecorry31, 2021-11-12:
- [ ] Upgrade to material 3 theme
- [ ] Verify UI

kylecorry31, 2021-11-12:
Blocked until material3 has a stable release

closed 2022-01-19 by kylecorry31

michaelblyons, 2022-01-26:
Can the bottom nav items keep their labels instead of bouncing up to show the label for the selected one?

kylecorry31, 2022-01-26:
@michaelblyons The Material Design guideline is to show only a single label if the labels are long. In English, for example, they are Navigation, Weather, Astronomy, Tools, and Settings - therefore overlap is likely and only a single label is shown.

Now if the Weather tab isn't available, I could probably fit all the English labels together - so maybe I could show the labels then if the screen is wide enough - I'll look into that, but won't be adding them when all 5 nav items are visible.