kylecorry31/Trail-Sense #1033: Don't reduce image size during calibration

kylecorry31, 2021-11-27:
Instead, record the calibration values (warp, rotation) and apply them before displaying the map

jptrzy, 2023-06-08:
Could you clarify what did you mean by writing the word **size**:
size in disk space or size in scale (how big image is).

kylecorry31, 2023-06-09:
@jptrzy I've updated the description of this issue to clarify. By size, I meant both disk space and resolution (though cropping will inherently reduce that).

StarTroop, 2025-06-24:
I'm wondering if this is relevant to the issue I'm having. I want to import a geospatial pdf, a fairly large (dimensionally) map which is sized 55mb. Whenever I import it, it always gets reduced to about 300kb, becoming unusably blurry. I've also tried making both larger and smaller versions of the same map (from the larger geotiff source, converted with gdal), which all get reduced in size to 300kb once imported.
I'm able to properly open and view these pdfs in my pdf viewer app. The practice geospatial pdfs I made with caltopo seemed to import fine, but they are much smaller to begin with.
Should I open this as a new ticket?

kylecorry31, 2025-06-24:
> I'm wondering if this is relevant to the issue I'm having. I want to import a geospatial pdf, a fairly large (dimensionally) map which is sized 55mb. Whenever I import it, it always gets reduced to [...]

Hi, have you tried turning off the reduce photo resolution setting in photo maps settings? If so, then yes I recommend logging a new issue using the bug template. 

That's assuming you aren't cropping the image during import (which is this issue)

StarTroop, 2025-06-24:
> 
> Hi, have you tried turning off the reduce photo/pdf resolution settings in photo maps settings? If so, then yes I recommend logging a new issue using the bug template.
> 
> That's assuming you aren't cropping the image during import (which is this issue)

Sorry, I didn't even realise there was such an option. I assumed this was a bug since the automatic reduction of 55mb to 300kb doesn't seem like a sensible default, IMO.
I've disabled the resizing but now the app crashes whenever I try to open my maps, though that's clearly a separate issue.

kylecorry31, 2025-06-24:
> Sorry, I didn't even realise there was such an option. I assumed this was a bug since the automatic reduction of 55mb to 300kb doesn't seem like a sensible default, IMO. I've disabled the resizing [...]

Yeah, that's a separate issue likely related to https://github.com/kylecorry31/Trail-Sense/issues/1345 

If you can provide some details in a new bug report that will help me investigate, though it's likely that the image is too large for the device.

I went with a high reduction by default to support lower end devices and for the primary use case of using the camera to import the map. That reduction amount was derived from my testing across various devices.