kylecorry31/Trail-Sense #1073: Harmonic CSV generator website

kylecorry31, 2021-12-23:
- [ ] Create a new website for generating tidal harmonics
- [ ] Search station IDs
- [ ] Poll NOAA API for harmonics given a station ID
- [ ] Convert harmonic JSON to CSV and download

This will only support the US for now, unless there are other free harmonic APIs for other countries.

This should be a separate repo.

Lee-Carre, 2022-01-01:
> Convert harmonic … and download

Given #1102; add the computed HC set to TS's source for future in-app bundling.

Lee-Carre, 2022-01-01:
> Convert harmonic … to CSV

See [#1072 (comment)](http://github.com/kylecorry31/Trail-Sense/issues/1072#issuecomment-1003611548).

kylecorry31, 2022-01-01:
I wouldn't imagine the bundling happens directly through the website - the website is designed to be a simple, client-side only website that lets the users download the harmonic file - I could include text / a link to email them to me or create a PR on GitHub though.

Lee-Carre, 2022-01-04:
> I could include text / a link to email them to me or create a PR on GitHub though.

That would do.

I'm largely indifferent about the specifics of the mechanism. The point is about the end result

If using harmonic predictions for tides isn't easy, then users will just stick to the simple tide clock.

A bundled set of HCs (because you can, unlike with tide clock references) makes harmonic prediction trivial (and clearly superior).

Users only need to deal with the technicalities when they wanna add or change something. Then it becomes a given that one must deal with technical details.

Merely using harmonic predictions should be trivial, a no-brainer, though.

kylecorry31, 2025-05-30:
With the implementation of the offline global tide model, I am no longer going to pursue this (it would take a lot of effort for limited use)