kylecorry31/Trail-Sense #1074: Tide analysis website

kylecorry31, 2021-12-23:
- [ ] Create a website to analyze tides (possibly the harmonic website)
- [ ] Upload tide levels for a station
- [ ] Calculate harmonics and download as CSV

Lee-Carre, 2021-12-27:
> Calculate harmonics

Given that the pipeline will likely end up being CPU-bound (else memory-bound): [Tidal Prediction With a Small Personal Computer](http://www.core.ac.uk/display/268172049).

Lee-Carre, 2022-01-01:
> Calculate harmonics and …

Given #1102, and that harmonic analysis isn't trivial; also add them to TS's source for future in-app bundling.

> download as CSV

See [#1072 (comment)](http://github.com/kylecorry31/Trail-Sense/issues/1072#issuecomment-1003611548).

Lee-Carre, 2022-01-01:
> Calculate harmonics and [output results]

A suggestion for a multi-stage approach. Nutshell: have 2 levels of processing.

First, do as originally intended; spit out the numbers which the user needs *now* in order to have harmonic predictions. Optimise for least-delay over highly-refined HC values. Even somewhat-rough numbers are likely to yield better predictions than the simple tide clock.
This processing should always have priority over what I'll describe next.

However, **keep the dataset**.

Separately, in the background at a lesser priority, perform more intensive processing on the (same) dataset, to produce a more refined set of HCs. This should be optimised for quality-of-results (for long-term accuracy of predictions) over the amount of time needed to compute them.
The results of this processing are what should be added to TS's repository of HC sets (for in-app bundling).
The submitted dataset may now be discarded.
A risk to account for, if the processing will take a long time (a week+ ?), is that the intermediate progress *not* be lost if the host is rebooted, loses power, or similar.

Using this strategy, users can avoid the trade-off between delay & quality, and have the best of both:
* rapid results which should be sufficient for the medium-term
* (later) results with a high confidence of accurate predictions into the long-term.

This strategy should be applied to each dataset submitted for processing.

kylecorry31, 2022-01-01:
I wouldn't imagine the bundling happens directly through the website - the analysis website is designed to be a simple, client-side only website that lets the users generate a harmonic file from water levels. Therefore the analysis would be one within the user's browser (i.e. there wouldn't be a server). I am planning on doing more research into how long this processing takes. I do like the idea of having multiple quality levels though.

Lee-Carre, 2022-01-04:
> the analysis website is designed to be a simple, client-side only …. Therefore the analysis would be one within the user's browser

Ah, in that case I'd rather use a CLI toolchain & DIY it.

In-browser wouldn't be that efficient, I imagine.

Makes me think that a [BOINC](http://boinc.berkeley.edu/) would be better, though very overkill.

Either way; some mechanism by which to avoid each user having to do his own harmonic analysis. It need only be done once; share the results (hence #1102).

kylecorry31, 2022-01-04:
I will consider also creating a CLI - I'll need to see how the website performs first / maybe consider using webassembly

kylecorry31, 2022-01-06:
After further consideration on who the users of this will be, I will be first creating a CLI that will be available here (https://github.com/kylecorry31/Trail-Sense-Tide-Analyzer)

kylecorry31, 2024-12-27:
Not going to implement.

- Frequency of use (ex. edge case) vs effort to implement

closed 2024-12-27 by kylecorry31