kylecorry31/Trail-Sense #1086: Handle scan intents for QR codes

kylecorry31, 2021-12-27:
- [ ] Allow the user to open TS to scan a QR code
- [ ] Allow the user to open TS to generate a QR code (maybe a share intent)

Lee-Carre, 2021-12-27:
> * … open TS to scan a QR code
> * … open TS to generate a QR code

While my thinking from #960 is still fresh; while I've no objection to being able to invoke TS for handling QR, I was seeking the inverse flow.

Being able to use something else to do the scanning, to pass the result back to TS (think similarly to delegating image-capture to a camera app).
Similar for generating a scannable code (good apps handle much more than just QR).

In my case, I'd want Binary Eye (or similar) to do the scanning (& history logging) of a QR (or any of the many code types it handles), and generating of codes I wish to send / share.
Binary Eye has a bunch of helpful features.

From TS's perspective, think of it as offloading the details of the task (just as for acquiring a photo). As opposed to reinventing the (entire) wheel.
Have a simple-mode which handles QR within TS, but another (expert / power-user) which then broadcasts intents to allow different tools to coöperate.
For cases involving uncommon code types, you then don't have to implement each of them; direct the user to enable intents and use a more specialised app. That code is then someone else's problem. Modularity, just on a larger scale.

kylecorry31, 2021-12-28:
Okay, that makes more sense. The current implementation should support this - Trail Sense only handles geo intents and should be picked up automatically by other qr scanners.

Similarly the QR codes generated by TS can be read by any QR scanner. Also the library I'm using to parse QR codes is the same as Binary Eye.

The main purpose of TS having a built in QR generator/scanner is to share data between devices without the internet - something that will be incorporated into more of TS' tools in the future. It can also be used to scan QR codes which may appear on trail signs or maps - and all those codes codes should use standard QR format (given TS' goals, I will not expand this to barcode or product codes, so the limited scope meets all the needs here)

With that said, I might reduce the scope of this issue to be around the generation of a QR code via a share intent.

kylecorry31, 2024-12-27:
Not going to implement.

- Frequency of use (ex. edge case) vs effort to implement

closed 2024-12-27 by kylecorry31