kylecorry31/Trail-Sense #1099: Tidal Data Sources

Lee-Carre, 2021-12-31:
Instead of cluttering up #1068:

* [NOAA § Tides & Currents § Harmonic Constituents](http://tidesandcurrents.noaa.gov/stations.html?type=Harmonic+Constituents)

Lee-Carre, 2022-01-01:
> In the US, constituents can be gathered [from NOAA]

I tried to find sources of tidal data for Britain, which yielded the following:
* [National Oceanography Centre](http://www.noc.ac.uk/facilities/data-research-facilities)
* [National Tidal and Sea Level Facility](http://www.ntslf.org/)

Annoyingly, it seems that not only aren't the harmonic constituents published, but even raw tide height measurements aren't available. Only the resulting predictions (for maybe 2 dozen days). (Mutters something unflattering & contemptuous about British attitudes toward data-hoarding, job-security, & control-freakery. Then something insulting about Ordinance Survey.) Of course, they're only too happy to sell licenses & consulting services, instead.

Seems that one is expected to both collect your own data, and then perform the harmonic analysis yourself.

What a joke, compared to NOAA. Sigh.

However, in my searching, I came across [CORE](http://www.core.ac.uk) (seemingly a British equivalent to ArXiv, hosting research papers), which hints at revealing more details in some of the documents.

Lee-Carre, 2022-01-01:
Australia's Bureau of Meteorology (being somewhat comparable to NOAA) has *some* tidal measurement data, at least:
* [National Tidal Unit](http://www.bom.gov.au/oceanography/projects/ntc/ntc.shtml)
* [Tide Gauge Metadata and Observed Monthly Sea Levels and Statistics](http://www.bom.gov.au/oceanography/projects/ntc/monthly/)

Though, seems to not be enough to do harmonic analysis.

Lee-Carre, 2022-01-01:
Did some searching for Jersey.

To my surprise, I found
[Jersey sea level and coastal conditions climate review](http://www.gov.je/sitecollectiondocuments/government%20and%20administration/r%20jersey%20sea%20level%20and%20coastal%20report%20final%20dm%2020170803.pdf)
which (on pages 36 & 37) gives (at least)
* M2 (0.392 metres)
* S2 (1.6 metres)

for Saint-Malo(, France).

Interestingly, references were made to
* [PSMSL](http://www.psmsl.org/)
* [SONEL](http://www.sonel.org/)

which I'm delving into now.

Lee-Carre, 2022-01-01:
> Annoyingly, it seems that not only aren't the harmonic constituents published, but even raw tide height measurements aren't available.

Seems that I may have spoken too soon: [British Oceanographic Data Centre](http://www.bodc.ac.uk/) 🙂.

I'll have to do more digging, but even if harmonic constituents aren't available, it seems like tidal data (sufficient to perform harmonic analysis) is.

Lee-Carre, 2022-01-01:
Digging reveals more clues, and possibly a goldmine of relevant data.

* [National Tidal and Sea Level Facility UK Tide Gauge Network 1990 - 2002 data](http://www.bodc.ac.uk/resources/products/data/bodc_products/ntslf/) (CD-ROM, lots of data, may be ordered for £0 🙂)
* [Sea level data](http://www.bodc.ac.uk/data/hosted_data_systems/sea_level/) (1980-to-present, not just British, either)
* [NW European Shelf Tidal Current Constituent Data Bank (1970-1988)](http://www.bodc.ac.uk/resources/inventories/edmed/report/581/): described as “The data set comprises full tidal analyses for over 800 current meter records collected at 400 sites in the seas around the British Isles, covering the continental shelf area and the shelf slope. The vast majority of the analyses in this data set are based on harmonic analyses, where the amplitude and phases of the tidal constituents are determined by a least squares fit. The data were selected from the BODC Current Meter Databank so as to provide representative coverage over the shelf areas — only good quality series were selected.”. While there's contact info for enquiries, the metadata does say “Availability: Licence; academic” but that may be more of ‘intended audience’ rather than exclusionary (🤞).
* in my reading, I encountered repeated references to World Ocean Circulation Experiment which involved a lot of data-sharing of (among other metrics) sea level (tidal) measurements. Apparently, part of the research was to compute harmonic constituents from the data. Part of [one description](http://www.bodc.ac.uk/projects/data_management/international/woce/tidal_constants/analysis/) read “For each station, the full multi-year time series was analysed. From this analysis the harmonic tidal constants (amplitude and phase lag for each tidal constituent) were determined.”

Lee-Carre, 2022-01-01:
[IOD Sea Level Station Monitoring Facility](http://www.ioc-sealevelmonitoring.org/) has a bunch of useful (meta)data:
* Sealevel Station Catalog (SSC): “The catalog is giving an overview of all identifiers used by the different data providers.”
* Direct data access web service: “Most of the station descriptive metadata and the station realtime data is accessible through a REST web service.”

Lee-Carre, 2022-01-01:
[MWTL physical monitoring network Westerschelde: Tidal Harmonic Constituents](http://www.vliz.be/en/find-datasets?module=dataset&dasid=1527)

Lee-Carre, 2022-01-01:
[Global Sea Level Observing System § data](http://www.gloss-sealevel.org/data)

Lee-Carre, 2022-01-01:
For PSMSL, their [High Frequency Data](http://www.psmsl.org/data/hf/) seems promising.

kylecorry31, 2022-01-01:
Thanks for all the information!