kylecorry31/Trail-Sense #1102: Tidal harmonic constituents; bundle, in-app, a known-good set

Lee-Carre, 2022-01-01:
Inspired by #1068.

> Have user enter the necessary constituents

Given #1099; related to #1073 & #549, supply an in-app list of (well-defined / stable / popular) locations, for convenience.

Ideally structure the source data (CSV(s)) such that Githubbers can easily add their favourite. By all means apply compression at build-time.

kylecorry31, 2022-01-09:
Things to consider for bundled tides:
- Popularity
- Proximity to a coastal trail (I don't want to add harmonics for areas outside of TS' use cases - unless interpolation can be done #1104 )
- Accuracy with the given HCs that TS will support

kylecorry31, 2024-11-24:
Added an experimental tide model that will be available in 6.6.0: #2518

closed 2024-11-24 by kylecorry31