kylecorry31/Trail-Sense #1105: Clock UI & UX review & suggestions

Lee-Carre, 2022-01-02:
* It seems to be showing time acquired from GNSS instead of the device's own clock; make this clear (it wasn't, to me, as an uninformed user). Besides clarity, it might give ideas for how it could be used. Before, I thought it was just an analogue dial showing the system time, which I thought was a little pointless. Now that I know it's showing (highly-accurate) time from GNSS (especially if I keep the receiver enabled), I have a bunch of uses for it 😀.
* Make clear which time standard it's using (UTC, TIA, UT, something else?).
* Display time elapsed since last sync (accounting for the case of the GNSS receiver being persistently-enabled, which I presume means that the elapsed time would always be zero(?)).
* Some kind of hint as to intended usage (this was also not clear to me as an uninformed user).
* Since one of the intended uses is to set the device's own internal clock, then displaying the offset of system time from UTC, would be helpful. If it's less than a second, I'm not fussed (because maintaining better than that is nigh-impossible, I find); more, and I'll want to correct the system clock.
* If supplied by the OS; estimate of the accuracy of the displayed (GNSS) time. As in, how much uncertainty is there in the displayed time (just like is done for position)? How does it likely match against UTC? Besides the likes of signal propagation-delay, this would (should) also include any factors on the device itself (e.g. software-induced, between the receiver chip and the display (with everything that's between).
* For the various metrics (above), by all means tuck them away in some `Info` dialogue or sub-menu. They don't *have* to be on the main UI.
* Kudos for implementing a second-hand which moves continually & smoothly. This is the ideal for analogue clocks (which only the very best do), rather than discrete steps. Time is continuous, by nature.

Those are all the simple ones, for now. I'll add more if I think of them. Non-trivial ones will be posted in separate issues.

kylecorry31, 2022-01-02:
Thanks, I may split this into smaller issues for when I pull this into a release cycle, but I agree with all your points - I also want to improve the look of the clock dial (I personally think the new clinometer dial looks good)

kylecorry31, 2026-03-14:
Replaced by the following:
- https://github.com/kylecorry31/Trail-Sense/issues/3576
- https://github.com/kylecorry31/Trail-Sense/issues/3577

closed 2026-03-14 by kylecorry31