kylecorry31/Trail-Sense #1110: Have the Tools menu sorted by most frecently used

Lee-Carre, 2022-01-04:
I often find that the tools I typically and/or commonly want (Tides, Clock) are several thumb-scrolls down. Would make for better UX that they rise to the top of the list.

Perhaps multiple sorting options (one of them being the current categorisation groupings), to cater to different preferences & work-flows.

Frecently isn't a typo; it's a portmanteau of frequently + recently. I'm not sure of the actual algorithm, though.
As examples to draw from; I do know that Mozilla Firefox implements it (for browsing-history purposes), as does the KISS launcher for Android.

kylecorry31, 2023-11-12:
In the next release you will have the ability to pin tools, I plan to add frequently used sorting in a future release

kylecorry31, 2024-12-27:
Not going to implement.

- Frequency of use (ex. edge case) vs effort to implement (user can pin the most frequently used tools)

closed 2024-12-27 by kylecorry31