kylecorry31/Trail-Sense #1111: Tide predictor: when there's only a single location to choose from, auto-selection shouldn't wait for precise position fix

Lee-Carre, 2022-01-04:
I happened to want to use the tide clock, while inside a building which prevents reception of GNSS signals.

TS tried to determine the position, I presume to match the nearest tidal location (as per my preferences). However, I have only a single location entry (being on a small island, there's no meaningful difference on any part of the coast).

If there were multiple entries; fair enough. It's doing what it's been told to do (display the nearest), which requires knowing roughly where I am.

However, when there's only 1 possibility for it to select? 😋

Unless it's a by-design (feature-not-a-bug) to sanity check that the user is at least *somewhat* near (within 1000 miles?) of the solo location, for it to be meaningfully relevant?

kylecorry31, 2022-01-04:
@Lee-Carre thanks for the suggestion, I'll do that. 

Also, I might need to update the description for require satellite fix to make that clearer - its intent is that you don't need at least 4 satellite fixes to use a location reported by GPS (and some devices might not provide that information)  (I can make a separate issue after)

closed 2022-01-08 by kylecorry31

Lee-Carre, 2022-01-09:
> need to update the description for require satellite fix to make that clearer — its intent is that you don't need at least 4 satellite fixes to use a location reported by GPS (and some devices might not provide that information) (I can make a separate issue after)

Ah, so minimum-quorum of visible satellites. Right. Makes more sense. Especially now that I recall the `passive` location provider isn't (yet) implemented.

So, yes, making the descriptive hint text clearer that it's about 3 vs. 4 satellites, rather than 4 vs. 0, would be good.

Thanks for the explanation, I'll re-enable that option, and wait for the one I thought it was 🙂.

Presuming that closure means ‘fixed’ (I didn't see the commit); thanks for that. Will test (when next able) and report back.

kylecorry31, 2022-01-09:
@Lee-Carre yes - closure in this repo means fixed/tested and on the main branch (but it doesn't necessarily mean released)