kylecorry31/Trail-Sense #1113: schedule and time mismatch

OmlineEditor, 2022-01-05:
the lines match for sunset and moonrise, but the time difference is 1 hour. please correct the graphs and time so that they match everywhere.

[image]

kylecorry31, 2022-01-05:
Hi, this might be an issue where your location isn't set up, so it doesn't know which location to calculate the times for (see the "GPS isn't working or sunrise/set times are inaccurate" of the FAQ: https://github.com/kylecorry31/Trail-Sense#faq) 

Let me know if that helps

I'll also double-check that the graph is displaying properly when the location is set up before closing this.

OmlineEditor, 2022-01-05:
after specifying the location and turning on the GPS, the problem is not solved

[image]

kylecorry31, 2022-01-05:
Okay, thank you - I will look into it.

kylecorry31, 2022-01-05:
Thanks for catching this - I was able to reproduce the issue (the horizon line isn't perfectly centered on the chart) and will fix it in the next release (this weekend). 

The times which are shown and altitudes when you tap the sun/moon icon are accurate though.

closed 2022-01-05 by kylecorry31