kylecorry31/Trail-Sense #1124: Support entry of tide tables

kylecorry31, 2022-01-08:
- [ ] Allow user to enter tide table times and heights
- [ ] Interpolate between highs/lows to display a chart (similar to what's done with the rule of twelfths) 
- [ ] Display the times the user entered
- [ ] (more research needed) calculate future times based on the times entered - and indicate they may be lower accuracy

kylecorry31, 2022-01-08:
The following algorithm can be used for creating the chart: https://github.com/kylecorry31/Trail-Sense-Tide-Analyzer/blob/main/tide_table.py

To avoid users skipping over a tide (or multiple), the interpolation should only happen if two entered tides are close in time to each other. 

Between specified tides, it might be useful to estimate the tides using either the M2 frequency, continuing the last sine wave, or using the average period for the high -> low or low -> high to estimate the next sine wave in the series.

closed 2022-01-10 by kylecorry31

Lee-Carre, 2022-01-25:
Review: been tinkering with this new implementation, and it works well. No major issues at all.

Since I noticed that you like to keep issues orderly, I'll open separate ones for the several minor suggestions I had.

I'm amused by the irony that, after our involved discussion re classical versus harmonic prediction, that my multi-reference-input idea ended up being implemented anyway 😋. Such is life.

Lee-Carre, 2022-01-25:
> Calculate future times based on the times entered

I have no idea what algorithms are being used, or their capabilities, but it did occur to me that it would be advantageous to leverage the astronomical info which TS already has, if only to know where in the lunar cycle an input reference relates to, and adjust accordingly.

Else, like you say, leveraging specific HCs that are available, for more accurate interpolation.

kylecorry31, 2022-01-25:
Essentially the algorithm is just the simple tide clock algorithm using the last high / low tide as a reference - so not super accurate, and definitely an area that can be improved (possibly with the lunar cycle idea).

And yeah, the multi-reference input idea ended up being super useful - thanks! I'm glad I did the use case analysis, because the new tide table algorithm allows the user to retrieve exact tide times for the ones they care about, but also lets them get some insight into future tides in case their trip is longer than planned. I had a lot of fun exploring the harmonic tides though and have all the algorithms in place for when I support some pre-packaged tides.

Lee-Carre, 2022-01-26:
> I'm glad I did the use case analysis

I wish more folks in the tech realm had this mindset. This is probably why TS is so loved by those who discover it.

Since you clearly care about usability & user experience, you may like to read [The Art of Unix Usability](http://www.catb.org/esr/writings/taouu/) (which seems to be a sequel to [The Art of Unix Programming](http://www.catb.org/esr/writings/taoup/)).

kylecorry31, 2022-01-26:
Thanks! I'll definitely give those a read