kylecorry31/Trail-Sense #1143: [Research] Auto detect thunder/lightning

kylecorry31, 2022-01-16:
While the lightning strike distance tool is open, it might be possible to do the following:

- [ ] Start the timer when lightning is detected with the camera
- [ ] Stop the timer when thunder is detected with the microphone

I don't know the feasibility of this, so this issue is to just research both of these. I might have to find footage during a thunderstorm, since I don't normally get many this time of year.

Lee-Carre, 2022-01-25:
Suggestion: use the direction in which the camera is pointed (presumably toward the storm) to feed weather-prediction.

A simple example being; is a storm in that direction (likely) approaching or receding?

kylecorry31, 2022-01-25:
Interesting, that would also relate to #102 (though using a different sensor / method)

Lee-Carre, 2022-01-26:
> Start the timer when lightning is detected with the camera

> I don't know the feasibility of this, …

While I don't know the technical details, I'm aware that photographers of lightning use an electronic device to trigger the shutter-release (since trying to do it by-hand would be far too slow).

I presume that such devices use the flash of bright white light as the trigger for their own mechanism.

A limiting factor might be the frame-rate of the camera system (60Hz?). Would it even see such a brief flash during preview mode? Particularly since the storm may be quite some distance away, and so the flash wouldn't fill the viewfinder frame (unless the digital zoom is used, I suppose).

I did think of suggesting the ambient light sensor, but I recall that they have a 5Hz sample-rate, which likely isn't enough. Even if they do averaging over the interval between samples, it's not like the change of intensity lasts long. The sensors being designed to determine the stable brightness of the environment, for merely auto-adjusting (roughly) the display brightness. So, they would probably ignore a brief flash, even if they detect it.

kylecorry31, 2022-01-26:
Yeah, I tried another project with the ambient light sensor for offline data transmission and it was far too slow. I read something about a lightning camera app on google play where the developer mentioned it being really tricky to detect lightning with the limited processing power/frame rate of Android phones. I still want to play around with this (even if it is just for fun) although I'm not hopeful that it will actually end up in TS 

For the accuracy required, having the user manually click buttons is fine though - but this would be pretty cool!

JensKorte, 2022-04-01:
* How about playing weather videos on TV and directing your phone on TV for testing purposes?
* how about doing an offline recognition with video recording and sound recognition and then starting to search for the lightning?
* how about starting the time count manually and stopping by sound recognition?
* how about starting video recording manually, marking the lightning manually, doing a sound recognition and searching for the lightning just within +-2 seconds of the lightning marker and otherwise use the marker.

kylecorry31, 2022-04-01:
Thanks for the suggestions, if I do this, I will try them out