kylecorry31/Trail-Sense #1144: Lightning strike distance history

kylecorry31, 2022-01-16:
- [ ] When the user gets a lightning strike distance, record the distance and current time
- [ ] Display the lighting strike history
- [ ] Delete readings older than 1 day
- [ ] Allow the user to delete readings manually

JensKorte, 2022-04-01:
Can the strike history be used for weather prediction?

kylecorry31, 2022-04-01:
Here are the things I think it could tell:
- If there is a storm
  - This wouldn't be useful though as you are using this tool because there is a storm (i.e. no added information)
- If the storm is moving closer or further from you
  - This would be useful as the speed and time of arrival/clearing up could be shown: https://github.com/kylecorry31/Trail-Sense/issues/1240 
- The intensity of the storm (frequency of lightning strikes - more research would be needed)
  - Similar to if there is a storm, this wouldn't really add any information the user isn't already aware of

closed 2023-01-07 by kylecorry31