kylecorry31/Trail-Sense #1171: remove unnecessary receivers BOOT_COMPLETED

OmlineEditor, 2022-01-30:
now there is a receiver BOOT_COMPLETED in the program.

this is an extra receiver and it interferes. at the same time, the program always starts when the phone restarts. please remove this resource or make it possible to disable it

[image]

kylecorry31, 2022-01-30:
I'll look into making this possible to disable - but I believe it is needed by Android work manager for the background processes that Trail Sense runs, and therefore I will not remove it.

Can you elaborate on how this interferes? I have not noticed any issues with this on my end.

OmlineEditor, 2022-01-30:
the problem is that I have little free RAM. when programs get into memory during a reboot, it becomes even smaller. this revisver is only for downloading the program when starting the device. it can be turned off without any problems.

kylecorry31, 2022-01-30:
Okay, I think I get it now - you want an option to disable TS starting on boot, which would mean to resume any of the background services, you would need to open Trail Sense after rebooting. Is that correct?

If so, that is something I can do

OmlineEditor, 2022-01-30:
you understood me correctly. please do this

commit 182831d7 (2022-02-13): Allow disabling start on boot

Closes #1171
closed 2022-02-13 by kylecorry31