kylecorry31/Trail-Sense #1177: Specify path width

kylecorry31, 2022-02-03:
Ex. the width of a river

This could be shown on the radar compass to convey how wide the path is. 

I'm not sure if I will implement this, more research is needed.

Lee-Carre, 2022-02-09:
Makes me think that (for trails) other properties might be useful; such as difficulty / quality metrics. Width being merely one of them.

I'm imagining hiking trails visited infrequently, and a user forgetting the differences. In the mind of the user, what they ultimately want is a reminder of which path they prefer.

kylecorry31, 2022-02-09:
The path tool already displays the difficulty and quality metrics - could you provide a specific example of what you would like added?

That sort of sounds like they should be able to provide a comment (like the beacon tool allows), here's a new issue to track that: https://github.com/kylecorry31/Trail-Sense/issues/1191

Lee-Carre, 2022-02-10:
> The path tool already displays the difficulty and quality metrics — could you provide a specific example of what you would like added?

Ah, my mistake for not checking / testing. More tinkering (on my part) needed.

Sounds like my imagined use-case is already covered.

> That sort of sounds like they should be able to provide a comment (like the beacon tool allows), here's a new issue to track that: #1191

Interesting; far more generic & adaptable. 👍

kylecorry31, 2026-11-02:
Not going to implement this due to complexity vs frequency of use

closed 2026-11-02 by kylecorry31