kylecorry31/Trail-Sense #1195: Cannot parse calibration data from GeoPDFs generated with QGIS

eresonance, 2022-02-12:
I downloaded a PDF from caltropo, opened it in TS and it worked perfectly, great!

However in my test file I generated with QGIS, using the following settings:
[image]
When I open it in TS it asks me to calibrate the file, so it seems parsing of the geopdf data is failing in some way.

I opened up the caltopo file in notepad++ and found the /GPTS line all nicely formatted:
```
8 0 obj
<<
/Type /Measure
/Subtype /GEO
/Bounds [0 1 0 0 1 0 1 1]
/LPTS [0 1 0 0 1 0 1 1]
/GPTS [45.057804 -75.91218 45.11057 -75.91218 45.11057 -75.851585 45.057804 -75.851585]
/GCS 22 0 R
/DCS 23 0 R
>>
endobj
```

However mine from QGIS has all this on one line:
```
6 0 obj
<< /GCS 7 0 R /GPTS [ 45.0780799698456534 -75.8550046364233737 45.0780799698456534 -75.8418074947041276 45.073006918482065 -75.8418074947041276 45.073006918482065 -75.8550046364233737 ] /LPTS [ 0 1 1 1 1 0 0 0 ] /Subtype /GEO /Type /Measure >>
endobj
```

I took a look at your commit:
https://github.com/kylecorry31/Trail-Sense/commit/53d7a63ba048a4f85ee435118ab72f6294eff71d#diff-4dc068e24ecb002befa7409a57987ed88a5c89e53fb1f26cfab4cd1ceae83e56R55

And I believe your regex in the linked commit is too greedy, so it pulls in those /LPTS whatever those happen to be:
[image]

However I butchered my pdf file a bunch attempting to get the /GPTS, /BBox, and /MediaBox lines to look similar to the caltropo one, and still couldn't get TS to import the map. It shows the map correctly but it always asks to calibrate it.

Anyways, I've attached the pdf file here for you to try, I don't quite see the issue. The pdf doesn't seem to have an actual raster map (not sure why, next thing for me to debug in QGIS!) but it still has all the geo info in it.

[test_modified.pdf](https://github.com/kylecorry31/Trail-Sense/files/8052609/test_modified.pdf)
[test_orig.pdf](https://github.com/kylecorry31/Trail-Sense/files/8052610/test_orig.pdf)

kylecorry31, 2022-02-12:
Thanks! That PDF definitely does not look to be in the format I expected. I figured everything would be in the correct order with proper new lines. 

I'll update the PDF parser to handle descriptors on the same line, which I believe will resolve this.

eresonance, 2022-02-12:
Awesome, thanks! I tried using a bunch of regex to force the <<, >>, and /FOO to be on separate lines but it still didn't import the coordinates. Maybe there's an error happening somewhere else in the PDF parsing :(

eresonance, 2022-02-12:
Also I made a small python script to get the coordinates from a geopdf for manual calibration, going to use your app for a camping trip today :)
https://gist.github.com/eresonance/3cd8840bd62db9159bd82eccd2e8a084

kylecorry31, 2022-02-12:
Okay, thanks for the details - it looks like there are some unexpected nested << >> sections around the "Mediabox" - I'll have to support that as well.

I'll experiment with this during my next development cycle to see if I can get it to work on the PDF you provided.

Hope you enjoy your camping trip!

kylecorry31, 2022-04-05:
Support for this will be added in 3.9.0. @eresonance I verified it auto-calibrates when I import the test_orig.pdf you uploaded (and I also tested with my own QGIS map) - thanks for providing that!

commit a9f0ff1b (2022-04-05): Add QGIS PDF support

Closes #1195
closed 2022-04-05 by kylecorry31