kylecorry31/Trail-Sense #1202: Add setting for front vs back camera

kylecorry31, 2022-02-14:
- [ ] Let the user choose to use the front or back camera
- [ ] Default to back camera

This is useful if the device does not have an operational back camera. Some tools must use the back camera though (Sighting Compass and potentially Clinometer)

AlessandroFrangiamone, 2024-02-14:
Hi @kylecorry31, i would like to help with this improvement.
Why not be able to provide the user directly with a flip camera button for tools where it makes sense to do so?
In case I have already developed a prototype since I liked the idea (obv the flip icon needs to be changed), I attach the video here:

https://github.com/kylecorry31/Trail-Sense/assets/34175360/d29d659b-5d0c-4802-a019-8967924ff266

kylecorry31, 2024-02-15:
Hi, thank you that improvement looks good. My only ask is that that functionality is not available on clinometer, sighting compass, bearing input, and augmented reality - since those were only developed with the rear camera in mind.

I will accept a PR for this, but I will likely hold off on merging until after the February release as I'm wrapping up my testing this weekend.

AlessandroFrangiamone, 2024-02-17:
Ok, thank you.
Before proceeding to create the PR of the specific behaviour on Trail-sense to include the change, I opened a PR required for integration with the Andromeda library: https://github.com/kylecorry31/andromeda/pull/93
The integration consists of exposing the `flipCamera` method for the `ICamera` interface and implementing this in the `Camera` class.

The improvement had to be made in the library because to change the camera it is necessary to unbind the lifecycle associated to the cameraProvider, a property not exposed by the library.

kylecorry31, 2024-02-17:
@AlessandroFrangiamone you can use Andromeda version 6e071b6b3a which contains your change (should finish building soon).

AlessandroFrangiamone, 2024-02-18:
Thanks! Regarding the icon to use for the camera flip, can I use this one: https://pictogrammers.com/library/mdi/icon/camera-flip-outline/ ? (If I'm not mistaken it should be the icon pack you have the licence)

kylecorry31, 2024-02-18:
Can you use the filled version? https://pictogrammers.com/library/mdi/icon/camera-flip/

AlessandroFrangiamone, 2024-02-18:
Ok, i opened the PR: https://github.com/kylecorry31/Trail-Sense/pull/2223

closed 2024-02-18 by kylecorry31