kylecorry31/Trail-Sense #1203: Offline file share companion app

kylecorry31, 2022-02-14:
Similar to Google Files offline share feature. Devices will share data over Bluetooth (either pair through Bluetooth, QR, or code entry).

This can appear as a custom share target for Trail Sense if installed and allow maps, paths, beacons, and settings to be transferred. 

Not a high priority - right now a user can export most of those to a file and then share the file (many Android devices will support Nearby share as well)

RealCreepi, 2023-01-19:
The most data dense method of bar codes are [HCCB](https://en.wikipedia.org/wiki/High_Capacity_Color_Barcode)s with 2 KB per square inch. On a Google Pixel 6 the transfer of the Survival Manual APK (5,5 MiB) would need 184 chunks. Automating the transmission of this payload by 'flipping' through chunks (see this [demo](https://canadaduane.github.io/elm-hccb/examples/streaming/index.html)) would take 73 seconds without any error correction codes in theory (with a delay time between chunks set to 400 ms). 

In the end, while highly impractical, I am sure that finding oneself in a situation where the usage of Bluetooth or WiFi Direct is undesireable isn't far too unlikely. In my opinion it should be bundled as a second option to the Bluetooth or WiFi Direct.

kylecorry31, 2023-01-19:
That was my main concern - and the devices would somehow need to communicate with each other to ask for the next code (could be done with light / sound - impractical for a person to do that, and error prone to have a fixed delay). I would also want error correction codes in place, since there are some pretty low quality cameras on low/mid range Android phones. And factoring in phones with low end processors, small screens, it could take quite a while. Though, it may have some practicality in transmitting smaller files, such as map images (a low resolution map in TS can be around 120KB)

I'll consider adding that as an option along side Bluetooth / WiFi direct - though all of those options can be used while offline (therefore they are more of a privacy concern rather than no internet access concern). In either case, I would have this as a companion application to Trail Sense, because I don't want TS to have Bluetooth or WiFi permissions.

JensKorte, 2023-02-11:
If you can make use of the front cameras two mobile phone could be faced to each other, one upside down and maybe with a stone below it to get the fitting height. The front cameras can record the display of the other mobile phone and you could get a full duplex connection.

JensKorte, 2023-02-11:
> (with a delay time between chunks set to 400 ms).

400ms would be 2.5 fps. If the phones don't need to decode the frames online but record a video I guess the speed could get up to at least half of the frame rate of the slowest camera, eg. the slowest camera can do 25 fps then 12,5 fps could  be possible. If you use 10fps that would result in  73 seconds/4= 18.25seconds. Maybe the data rate could be doubled by showing 2 QR codes or even more at once.

> In my opinion it should be bundled as a second option to the Bluetooth or WiFi Direct.

+1

kylecorry31, 2023-02-11:
I hadn't thought of the two devices facing each other and communicating via the screen. That definitely makes this more feasible, as both users could just hold their phones facing each other for a little bit, and a time estimate/progress bar can be shown.

kylecorry31, 2024-03-02:
Closing in favor of https://github.com/kylecorry31/Trail-Sense/issues/1540 (wifi/bluetooth/NFC communication) and https://github.com/kylecorry31/Trail-Sense/issues/1088 (QR communication)

closed 2024-03-02 by kylecorry31