kylecorry31/Trail-Sense #1322: White Noise feature not working

napsolo, 2022-06-06:
Hey,
I downloaded the app via fdroid store and found out that, White Noise feature is not working on my mobile, when turned on it makes a sound and nothing after that. Kindly look into this issue.

Thanks.

kylecorry31, 2022-06-06:
Hi, could you provide the following information so I can try to reproduce the issue (it is working fine on my device):

1. Android version
2. What noise does it make when you turn it on?
3. Does a notification appear when you turn it on?
4. Are you turning it on from the White Noise tool or via a quick action on another tool (ex. Navigation, Astronomy, Weather)?
5. Is the sleep timer set on the White Noise tool, if so what is it set on?

kylecorry31, 2022-07-10:
@napsolo I believe this should be fixed in the next release (4.1.0) - the way I could reproduce it was to set the sleep timer and let it run down to 0 and then try to run the white noise again from the white noise tool. If you were experiencing this in other cases or are still experiencing it after version 4.1.0 is released, please let me know and respond with the information requested in my above comment and I can reopen the issue.

commit 9ea8c3b7 (2022-07-10): Fix white noise after sleep timer set

Closes #1322
closed 2022-07-10 by kylecorry31