kylecorry31/Trail-Sense #1347: Label compass with magnetic vs true

gdt, 2022-06-28:
I know that there is a setting for magnetic vs true, but I returned to Trail Sense after long enough that I didn't remember how I had set it.  It would be nice if  the compass readout showed "335\deg T  NW" or "335\deg M  NW" depending on the option.

kylecorry31, 2022-06-28:
Good idea - I'll experiment with the text / icon to see what works best.

kylecorry31, 2023-10-15:
Also add this badge to the bearing input view

kylecorry31, 2023-11-05:
Maybe show an icon (magnet vs globe), when tapped shows that it is magnetic or True North. 

I don't think abbreviations will work (would be confusing and not great for localization), and the text is too long to show by itself.

gdt, 2023-11-05:
Magnet vs globe makes sense.   The UI style of an icon and then a help method to get a proper, not contrained to be too short, localized text sounds great to me.

kylecorry31, 2023-11-07:
Tasks:

- [x] Show north reference on navigation tool
- [ ] Show north reference on bearing input view
- [ ] When north reference is clicked on the navigation tool, show a dialog about what is being used and link to settings to change it

commit 1932a823 (2023-11-18): Add north reference description

Closes #1347
closed 2023-11-18 by kylecorry31