kylecorry31/Trail-Sense #1423: Bubble level accuracy

JonnyTech, 2022-08-15:
Is is possible to display a greater accuracy with bubble level similar to other apps?

[image]

[image]

JonnyTech, 2022-08-15:
Thank you for your consideration. I am using it to show the kids how to build booby traps for protection when outdoors so it is kinda within the use case ;)

kylecorry31, 2022-08-16:
Actually, it doesn't hurt to add it and it is pretty easy to add now, so I'll just do that for the next release (I already have changes to the bubble level in the release, so it won't add much extra testing)

closed 2022-08-16 by kylecorry31

JonnyTech, 2022-08-16:
Brilliant - thank you!