kylecorry31/Trail-Sense #1457: an interesting old app

vinniec, 2022-09-08:
when I saw your app I immediately thought of an old app which unfortunately no longer works on new devices but which could give you a good idea for an additional tool. 

https://f-droid.org/en/packages/de.hu_berlin.informatik.spws2014.mapever/

 It simply allows you to create an amorphous gps map based on the points entered, so it allows for example to navigate on a hand drawn map with all the wrong dimensions.  When it worked it was great.

kylecorry31, 2022-09-08:
Thank you for the suggestion! I actually received a recommendation to take a look at this app shortly after Trail Sense's offline maps were created - unfortunately, I couldn't run it on my device either, but I looked at the source code for it and saw it uses a different algorithm than I do to calibrate map images. I created the following issue to investigate using a similar algorithm: https://github.com/kylecorry31/Trail-Sense/issues/1330 (this issue should now be linked there for reference)

closed 2022-09-08 by kylecorry31

vinniec, 2022-09-08:
thanks for the appreciation