kylecorry31/Trail-Sense #1465: Error upon opening

imBigo, 2022-09-14:
I'm not sure what happened but this error showed up. I didn't notice any issue but the error message since I had only started the app.

Here is the log:
```
Trail Sense 4.4.0
Android SDK: 33
Device: Google redfin (Pixel 5)
Fragment: LevelFragment
Diagnostics:
Message: Context.startForegroundService() did not then call Service.startForeground(): ServiceRecord{f9b454 u0 com.kylecorry.trail_sense/.tools.flashlight.infrastructure.FlashlightService}

android.app.RemoteServiceException$ForegroundServiceDidNotStartInTimeException: Context.startForegroundService() did not then call Service.startForeground(): ServiceRecord{f9b454 u0 com.kylecorry.trail_sense/.tools.flashlight.infrastructure.FlashlightService}
at android.app.ActivityThread.generateForegroundServiceDidNotStartInTimeException(ActivityThread.java:2006)
at android.app.ActivityThread.throwRemoteServiceException(ActivityThread.java:1977)
at android.app.ActivityThread.-$$Nest$mthrowRemoteServiceException(Unknown Source:0)
at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2242)
at android.os.Handler.dispatchMessage(Handler.java:106)
at android.os.Looper.loopOnce(Looper.java:201)
at android.os.Looper.loop(Looper.java:288)
at android.app.ActivityThread.main(ActivityThread.java:7898)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
```

imBigo, 2022-09-14:
This is probably similar to #1458 since I am still using version 4.4.0.

Also, I'm wondering if this issue could be related to my phone recently having its flashlight randomly turn on?

kylecorry31, 2022-09-14:
Hi, I believe that is the same issue as #1458, so should be fixed in version 4.5.0. 

Trail Sense has the flashlight bound to the volume key (toggleable in settings, but I believe it is off by default if I remember correctly) - is it possible you pressed the volume button while Trail Sense was open (assuming it is enabled)?

Alternatively, if something else turned on your flashlight (ex. a phone gesture - I used to have a Motorola phone which if you did a chopping motion would turn on the flashlight), Trail Sense will start up its flashlight service when something else opens the flashlight (so you can turn it off via the notification - this is toggleable in settings, default to enabled)

The error message indicates it occurred while you were using the level - there are no flashlight integrations on that tool (other than the volume key and system toggle mentioned above)

imBigo, 2022-09-14:
Hi, I have a Pixel 5 and I did have the "Quick Tap" (where you double tap the back of the phone) feature turn on my flashlight before. However, I have since then turned it off and that is when the issue started happening. 

It would happen when Trail Sense was not in the foreground. Now it seems to be gone.

I will start using 4.5.0 and if the bug still happens I will reopen the issue.

Thanks

closed 2022-09-14 by imBigo