kylecorry31/Trail-Sense #1467: Research Wear OS support

kylecorry31, 2022-09-17:
- [ ] Determine steps needed
- [ ] Determine which features would make sense to port

rmnwnsx, 2022-10-21:
As you said, there are some people, who want this feature, and you need time and device to port and test Wear OS app. Maybe you can make a goal or tier in github sponsors specifically for porting to wear os and this people can donate to you.

kylecorry31, 2022-10-21:
@rmnwnsx thank you, that is a good idea - I have several major features which I want to ensure are finished first within Trail Sense before I commit to adding a goal/tier for that. But once those are done and I confirm it is feasible to do, I can create a tier.

The major feature roadmap (with tentative dates) is outlined here: https://github.com/kylecorry31/Trail-Sense#feature-roadmap

kylecorry31, 2024-12-27:
Unfortunately, I will not be adding this. The time and effort is too much for me to make the port and I don't have a smart watch (nor do I plan to get one)

closed 2024-12-27 by kylecorry31