kylecorry31/Trail-Sense #1477: The compass is not readable - too small

panwanarek, 2022-09-21:
It used to be OK, but now I can't even read where midnight is at 1080 x 2340 display resolution
[https://i.postimg.cc/DfRhjdX7/Screenshot-20220921-150352-Trail-Sense.png](url)

kylecorry31, 2022-09-21:
@panwanarek Hi, I'm not sure I see the issue - the image you posted appears to be what the radar compass has always looked like and there have been no changes to the size of that recently. If you are looking to switch to the standard compass, you can open Settings > Navigation > Disable Show nearby radar compass (I'm not sure if that's what you mean, but that is a larger compass).

If that is not what you are looking for, can you clarify what you mean about not being able to read where midnight is and when this issue started appearing?

panwanarek, 2022-09-21:
"Disable Show nearby radar compass" makes the komaps slightly bigger. But it's still unreadable to me. The orange midnight sign (in the sun on the phone screen) is completely unreadable on the black background. Maybe at least an arrow that signifies the north? The best version with a white dial. Please

kylecorry31, 2022-09-21:
Thanks - the cardinal direction text is designed to be small on the radar compass since if they were too big it would conflict with the beacons and paths displayed on the compass - I'll look into making them a little bigger and adding an icon over North though.

As for the white dial, I created this ticket to track changing the background color of the compass: https://github.com/kylecorry31/Trail-Sense/issues/1478

panwanarek, 2022-09-21:
Thank you very much for the information and reaction
I'll just show you the difference between the compasses. Left: very easy to read, right: poorly. I chose any example from Appsore. Trial sense has much more possibilities, only the main compass is poorly readable.

[image]

kylecorry31, 2022-10-01:
(my implementation checklist for planning purposes)

- [ ] Increase size of cardinal directions
- [ ] Make North symbol white
- [ ] Add arrow above North (experiment with how this looks / if it is confusing with beacon navigation)

kylecorry31, 2022-10-01:
There will be changes to this in 4.7.0

The text size is slightly larger on the radar compass (I couldn't go much larger without it cluttering up the compass when beacons/paths are shown) and the North symbol is white to match the other directions.

closed 2022-10-01 by kylecorry31