kylecorry31/Trail-Sense #1478: Custom compass background color

kylecorry31, 2022-09-21:
Allow the user to choose the background color of the compass for optimal readability.

Portagoras, 2023-02-19:
I am currently working on this and got a first implementation. I used the Color Picker that was already in the app. It is insufficient I think, as it has mostly bright colors available, which conflicts with the rest of the compass design.

Also there is no way to go back to default with that.

Maybe let me know what you think of this. I pushed it to a branch on my fork, if you wanna take a deeper look.

[image]
[image]

[image]

kylecorry31, 2023-02-19:
Hi, thank you for taking a look at this issue.

A few things:
1. The colors in that picker were chosen to be visible on desaturated surfaces, and therefore should be limited to just the beacons/paths.
2. I planned on doing more research before this was ready to see what colors would work, unfortunately I don't have the time at this moment to create a list. But a good starting point would be to only allow the selection of desaturated colors such as dark grays/black or light grays/white. This issue was logged out of feedback that a user was having trouble seeing the compass in direct sunlight and wanted a more contrasting background for their screen (white background).
3. For the colors for this, I would extend AppColors to have a property for isForegroundColor, then update the Color Picker to filter based on it, so you can specify if you want to see foreground or background colors.
4. The color of the text and outlines on the compass itself should be easily visible (switching between black and white based on the most contrasting color - there's a custom method, something like Colors.mostContrastingColor for that)
5. As long as the current color is in the list of AppColors and the preference defaults to use that one, I wouldn't worry about the ability to revert to default - the user could always just pick it again
6. The setting should be on the Navigation > General settings rather than the compass settings and apply to both the RoundCompass and RadarCompass

Portagoras, 2023-02-19:
Thanks for the feedback, I will look into it next weekend!