kylecorry31/Trail-Sense #1486: Sunset alert status bar

kylecorry31, 2022-09-26:
Similar to the status bar on the weather tool, show the sunset alert status on the astronomy tool and let the user toggle it from there.

aw-bib, 2022-11-28:
In context it might be sensible to replace a _sunset tool bar_ by a _moon rise bar_. Or "the next astronomical event" if it's not about having light.

From a user perspective mind that the current layout for astronomical events is logical and nice, but uses a lot of white space. Adding another bar there (especially the size like the one on the weather tab) will eat quite a bit of valuable space and require even more scrolling. So maybe one should also consider a more compact (table) layout. E.g. one line `Rise | Noon | Set` per body or the like. Or allow to disable it. (On the weather tab I'd love to disable the bar especially as barometric tracking is on for me all the time.)

kylecorry31, 2022-11-28:
@aw-bib thanks, my main goal of this is to indicate that the sunset alert is enabled and provide a quick way to toggle it. But I'll look into what it would be like if I merged this with the astronomy alerts as well. I feel that the sunset alert is more likely to be toggled than the other alerts, which is why I want that to be the main focus. Another option could be to provide a quick action toggle (top corners) that can be swapped out (same for weather monitor).

I've created another ticket to track the visibility of the status bars: https://github.com/kylecorry31/Trail-Sense/issues/1574 and another one for a compact astronomy view https://github.com/kylecorry31/Trail-Sense/issues/1575

aw-bib, 2022-11-29:
> my main goal of this is to indicate that the sunset alert is enabled and provide a quick way to toggle it. 

Ah. I miss understood the point.  I was more thinking of a prominent display of "next event where I get/loose light".

I wonder why the toggles (same for weather) should be _that_ prominent, but in view of 

> I feel that the sunset alert is more likely to be toggled than the other alerts,

this might be plausible. I'd still suggest smaller simple buttons similar and I like

> Another option could be to provide a quick action toggle (top corners) that can be swapped out (same for weather monitor).

> I've created another ticket to track the visibility of the status bars

Thx.

kylecorry31, 2022-11-29:
I can probably make them a bit smaller - my reasoning to make them so prominent was to reduce the number of support tickets I was getting around "why am I receiving this notification" or "how do I turn X off" - I figured having them there would make it much clearer that for example the "Weather Monitor" feature is running and can be easily stopped. Based on the reduction in support tickets around weather monitor and backtrack, it appears to have done the job (might be correlation rather than causation though, I'll give it a bit longer before saying for sure it helped).

But I agree it makes sense to have a way to hide them for more tech savvy or always on users.

kylecorry31, 2022-11-29:
I'm not sure about a way to add a settings button to each module - I was wondering the same thing about the user guide, maybe via a gesture, but that seems hard to discover.

I'll try to find some middle ground when I make this change (I test on a nexus 4 emulator, and I do realize that the astronomy tab is getting cramped - much different than the experience on my Pixel 5A with a tall screen)

aw-bib, 2022-12-01:
> maybe via a gesture, but that seems hard to discover.

Indeed this is nice but hard to discover.

I was wondering if some burger menu in the usual upper right would work. OTOH right now there are no burger menus in TrialSense at all... So it would be quite a change to add the way to (the proper) settings from each module there. One could imagine a menu item for the settings and one for the relevant sensors as a start and add the switches envisioned. Hm. Quite a change.

Another idea would be the setting/switch as a quick action configured as default action. Once you know how to change it you probably know you way around the deeper parts.

Probably nothing is ideal.

kylecorry31, 2023-09-04:
Not going to add this for now - replaced by the quick action for sunset alerts

closed 2023-09-04 by kylecorry31