kylecorry31/Trail-Sense #1505: Scan QR code from image

kylecorry31, 2022-10-10:
Allow an image to be scanned rather than just from the camera.

7eventech77, 2024-02-19:
+1 Would like to have this feature as well

AlessandroFrangiamone, 2024-02-24:
Hi @kylecorry31, @7eventech77, what do you think, would that be ok?

https://github.com/kylecorry31/Trail-Sense/assets/34175360/7b19ee4a-3792-408e-b1f7-6e52899411b1

@kylecorry31 a couple of doubts I have about the implementation:

- In the event that the qr code is not recognised, do i include a message of some kind e.g. a Toast/snackbar saying 'the qr code is not readable' or something like that?

- I used the `ExifInterface` class to allow the rotation of a bitmap, specifically `android.media.ExifInterface`. This is supported by API 24, as the project has minsdk 23 I have to import the support library for ExifInterface (`com.android.support:exifinterface`). Is this approach ok for you?

kylecorry31, 2024-02-24:
> Hi @kylecorry31, @7eventech77, what do you think, would that be ok?
> 
>  Screen_recording_20240224_223349.mp4 
> @kylecorry31 a couple of doubts I have about the implementation:
> 
> * In the event that the qr code is not recognised, do i include a message of some kind e.g. a Toast/snackbar saying 'the qr code is not readable' or something like that?
> * I used the `ExifInterface` class to allow the rotation of a bitmap, specifically `android.media.ExifInterface`. This is supported by API 24, as the project has minsdk 23 I have to import in gradle the support library for ExifInterface (`com.android.support:exifinterface`). Is this approach ok for you?

Thank you, that looks good. I have a few comments:
1. You shouldn't need to rotate the image, as the QR code scanner should be rotation invariant (that should reduce the memory footprint / remove the need for the library)
2. When you read the contents of the image file, be sure that you are downscaling the image (max 2048x2048 preferably, and if possible smaller as some devices have memory limitations).
3. I would have the message say "No QR code detected" and use a toast (there's a toast extension method on the AndromedaFragment)
4. Is there enough room on small screen devices with the button placement there (I typically use the Nexus 4 emulator)? If not, I would suggest maybe using a floating action button with padding at the bottom of the results list so you can scroll fully (but I do prefer the button as you have it for clarity)
5. Finally, can you add a loading indicator after the user selects the image while it scans in case it takes a while (ex. on a lower end device)?

7eventech77, 2024-02-25:
> Hi @kylecorry31, @7eventech77, what do you think, would that be ok?
>
>  Screen_recording_20240224_223349.mp4
> @kylecorry31 a couple of doubts I have about the implementation:
>
> * In the event that the [...]

Yes, that implementation is good from my perspective including the the comments @kylecorry31 made on it. The only suggestion/questions I have are below.

Does your implementation work with WiFi network QR codes or any other types other than URLs? I know that one of the reasons I would like to have this feature is to be able to read a Wireless network QR code that Android allows you to save as an image. The password is not printed in text on the QR code so without a tool to scan the QR and give the Network Name/Password one would not be able to retrieve the password to a saved network. The ability to read Geo coordinates I could see also being useful when you are out in the field if there are QR codes with them on trails. The ability to read other types of QR codes such as text, email, SMS, Phone, etc. would be useful as well.

Except for that looks great, thank you!

AlessandroFrangiamone, 2024-02-25:
> Thank you, that looks good. I have a few comments:
>
> 1. You shouldn't need to rotate the image, as the QR code scanner should be rotation invariant (that should reduce the memory footprint / remove [...]

Thank you, i have completed/checked all points, here is the PR: https://github.com/kylecorry31/Trail-Sense/pull/2240

kylecorry31, 2024-02-25:
> Yes, that implementation is good from my perspective including the the comments @kylecorry31 made on it. The only suggestion/questions I have are below.
>
> Does your implementation work with WiFi [...]

It can read wifi QR codes (and any other type of QR code), but it doesn't actually support doing anything with them other than seeing the text the QR code contains. I don't plan on adding support for actually connecting to the wifi network in Trail Sense though. URLs are also supported, and you choose to open them in your phone's browser. As for geo QR codes, that feature is also already available - and it shows options such as creating a beacon at the location shown.

For more information you can see this guide on the QR Code Scanner tool: https://kylecorry.com/Trail-Sense/guide?id=tool_qr_code_scanner

7eventech77, 2024-02-25:
> > > Hi @kylecorry31, @7eventech77, what do you think, would that be ok?
> > > Screen_recording_20240224_223349.mp4
> > > @kylecorry31 a couple of doubts I have about the implementation:
> > > 
> > > * In the event that the qr code is not recognised, do i include a message of some kind e.g. a Toast/snackbar saying 'the qr code is not readable' or something like that?
> > > * I used the `ExifInterface` class to allow the rotation of a bitmap, specifically `android.media.ExifInterface`. This is supported by API 24, as the project has minsdk 23 I have to import in gradle the support library for ExifInterface (`com.android.support:exifinterface`). Is this approach ok for you?
> > 
> > 
> > Yes, that implementation is good from my perspective including the the comments @kylecorry31 made on it. The only suggestion/questions I have are below.
> > Does your implementation work with WiFi network QR codes or any other types other than URLs? I know that one of the reasons I would like to have this feature is to be able to read a Wireless network QR code that Android allows you to save as an image. The password is not printed in text on the QR code so without a tool to scan the QR and give the Network Name/Password one would not be able to retrieve the password to a saved network. The ability to read Geo coordinates I could see also being useful when you are out in the field if there are QR codes with them on trails. The ability to read other types of QR codes such as text, email, SMS, Phone, etc. would be useful as well.
> > Except for that looks great, thank you!
> 
> @7eventech77 It can read wifi QR codes (and any other type of QR code), but it doesn't actually support doing anything with them other than seeing the text the QR code contains. I don't plan on adding support for actually connecting to the wifi network in Trail Sense though. URLs are also supported, and you choose to open them in your phone's browser. As for geo QR codes, that feature is also already available - and it shows options such as creating a beacon at the location shown.
> 
> For more information you can see this guide on the QR Code Scanner tool: https://kylecorry.com/Trail-Sense/guide?id=tool_qr_code_scanner

Ah okay. I completely understand, being able to view the text of the WiFi QR code is enough (aka the password mainly), just wanted that since it is such a pain to be able to get the password of a network you have saved but forgot the password of. Thank you!

kylecorry31, 2024-03-07:
@AlessandroFrangiamone Hi, I was doing some testing of this and found that it isn't detecting most images of QR codes. I tried to update the QR library to do a high accuracy scan, but that didn't seem to help much. After some more digging, I found that zxing (the underlying library) is designed primarily for mobile video and prioritizes speed over accuracy, which is why it seems to be missing a lot of QR codes. 

Unfortunately, I don't have more time to look into improving this right now, so I've hidden the button from the UI (to avoid getting bug reports). If you have time, here's where the QR code scanning is being done - it may be possible to set additional flags or similar to make it more accurate while scanning in high accuracy mode: https://github.com/kylecorry31/andromeda/tree/main/qr

kylecorry31, 2026-02-22:
Due to the technical limitations of the QR library, I will not be implementing this